gnuradio/gnuradio #5406: New release for python3.10

blackwiz4rd, 2021-12-18:
### Feature Description

Hi, due to the issue in #5401 I am unable to make gnuradio work on Arch Linux with python 3.10 installed. 
Since it is not possible to downgrade python's version, I installed python 3.9 with `pacman -S gnuradio39`, however when executing the script it cannot find gnuradio's module. I am not sure of the reason behind this (probably a mismatch of modules and python versions):
`ModuleNotFoundError: No module named 'gnuradio'`
Is there any suggestion to quickly fix this?
It would be easier if you release gnuradio 3.10 so that I can use gnuradio again?
Thanks a lot.

### Feature Urgency

high (impacts broad portion of ecosystem)

marcusmueller, 2021-12-18:
How did you install / build GNU Radio, and which version / branch ?

closed 2021-12-18 by marcusmueller

blackwiz4rd, 2021-12-20:
> > I installed python 3.9 with `pacman -S gnuradio39`
> 
> do you mean
> 
> > I installed python 3.9 with `pacman -S python39`
> 
> ? If so, how did you afterwards build GNU Radio? From which version / branch of the sources? As Jeff said, the fix for this is already on both the `maint-3.9` and the `maint-3.10` branches, so this _has_ been fixed, it's just not made it into a release yet (and we're currently in the process of releasing).

Yes, I meant `pacman -S python39`, sorry for the typo. I installed gnuradio with `pacman -S gnuradio` (https://archlinux.org/packages/community/x86_64/gnuradio/, gnuradio 3.9.4.0-2)

blackwiz4rd, 2021-12-20:
> Yes, I meant `pacman -S python39`, sorry for the typo. I installed gnuradio with `pacman -S gnuradio` (https://archlinux.org/packages/community/x86_64/gnuradio/, gnuradio 3.9.4.0-2)

> do you mean
>
>
> ? If so, how did you afterwards build GNU Radio? From which version / branch of the sources? As Jeff said, the fix for this is already on both the `maint-3.9` and the `maint-3.10` [...]

Installing `maint-3.10` from source solved the issues. Sorry for bothering with this ghost issue.