gnuradio/gnuradio #5407: Doxygen: obsolescence warning about COLS_IN_ALPHA_INDEX

marcusmueller, 2021-12-19:
### What happened?

When building maint-3.9 (and probably also maint-3.10/master) doxygen 1.9.1 complains about this setting being obsolete.

Since the number of columns in the alphabetic index probably isn't something to lose too much sleep about, this warning should be silenced.

### System Information

OS: (Linux Distro, Mac, Windows)
GR Installation Method: (native distro packages - apt, rpm, etc., Conda, Source

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

v3.9.4.0-40-g823f73013

### Steps to Reproduce the Problem

```
git checkout maint-3.9
git fetch
git reset --hard origin/maint-3.9
cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_INSTALL_PREFIX=$GRPREFIX -GNinja -DCMAKE_CXX_COMPILER_LAUNCHER=ccache -DCMAKE_CXX_COMPILER=clang++  -DCMAKE_C_COMPILER=clang -DCMAKE_MODULE_LINKER_FLAGS=-fuse-ld=lld  -DCMAKE_SHARED_LINKER_FLAGS=-fuse-ld=lld -DENABLE_CTRLPORT_THRIFT=Off  -Wno-dev  -DENABLE_GR_CTRLPORT=ON -DENABLE_GNURADIO_RUNTIME=ON -DENABLE_PERFORMANCE_COUNTERS=Off ..   && ninja
```

### Relevant log output

```shell
warning: Tag 'COLS_IN_ALPHA_INDEX' at line 968 of file '/home/marcus/src/gnuradio/build-3.9/docs/doxygen/Doxyfile' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
```

drmpeg, 2021-12-19:
Fixed on master/maint-3/10 with https://github.com/gnuradio/gnuradio/pull/5381 Just needs to be backported.

willcode, 2021-12-19:
I forget why we decided not to backport this originally. Let's give it a try ...

PR #5408 (2021-12-19): doxygen: Remove obsolete COLS_IN_ALPHA_INDEX for doxygen 1.9 and later (backport to maint-3.9)
Backport https://github.com/gnuradio/gnuradio/pull/5381

Fixes #5407
closed 2021-12-27 by marcusmueller