gnuradio/gnuradio #5411: v3.8 does not build on (my) Fedora 35

willcode, 2021-12-19:
### What happened?

Building maint-3.8 on Fedora 35 fails at ` Swig source vocoder_swig.i`

### System Information

OS: Fedora 35
GR Installation Method: Source (PyBOMBS)

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

maint-3.8 (at v3.8.5.0-rc1)

### Steps to Reproduce the Problem

Build

### Relevant log output

```shell
[ 97%] Swig source vocoder_swig.i
/usr/include/codec2/freedv_api.h:148: Error: Syntax error in input(1).
make[2]: *** [gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/build.make:77: gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp] Error 1
make[2]: *** Deleting file 'gr-vocoder/swig/CMakeFiles/vocoder_swig.dir/vocoder_swigPYTHON.stamp'
make[1]: *** [CMakeFiles/Makefile2:10181: gr-vocoder/swig/CMakeFiles/vocoder_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```
```

willcode, 2021-12-19:
I don't see any related changes in maint-3.8, all the way back to the previous release. CI builds are good. I'm pretty sure I got a good build off a recent version. Mystery.

willcode, 2021-12-19:
F35 upgraded codec2 from 1.0.0 to 1.0.1 recently. Possibly related.

willcode, 2021-12-20:
According to Maitland, there is something that needs fixing with codec2 v1.0.1, and Debian patches it.

marcusmueller, 2021-12-26:
Can confirm that fedora35 codec2 is unpatched; but line 148 in that file is:
https://github.com/drowe67/codec2/blob/v1.0.1/src/freedv_api.h#L148:

```c
typedef void (*freedv_callback_rx_sym)(void *, _Complex float, float);
```

Don't think that's a syntax error (it *is* C in its maximum beauty, though).

So, being on a train with bad internet, can't run a F35 container build locally. I'll work under the "SWIG being weird *again*" hypothesis.

So, foo.h:

```c
#include <complex.h>
typedef void (*freedv_callback_rx_sym)(void *, _Complex float, float);
```
```shell
swig -python /tmp/foo.h
```
indeed yields 
`/tmp/foo.h:2: Error: Syntax error in input(1).`; hmmmm.

marcusmueller, 2021-12-26:
errr cdecl.org says it *is* a syntax error, gcc disagrees... sigh

marcusmueller, 2021-12-26:
Ah, it's a syntax error in C89 because that doesn't know `_Complex`; I can make my testcase's failure go away using 
`swig -python -D_Complex=complex /tmp/foo`. The reason is that SWIG emulates C99 complex number support incorrectly; instead of having a define in a header that defines `complex` as `_Complex` and the latter being a C99 keyword.

`_Complex` is a C99 keyword, `complex` isn't, but a macro in `<complex.h>`. SWIG decided to interpret `complex` as keyword, and be ignorant about `_Complex`.  Well. We can (theoretically circularly) define `_Complex` to mean `complex`.

marcusmueller, 2021-12-26:
See the untested workaround above. I wish we could just `%ignore` the problematic parts of the API (our intent is definitely not to expose the full codec2 API to Python), but this already fails when SWIG is parsing the header.

willcode, 2021-12-26:
Same error with or without this patch. I didn't track it down any further.

marcusmueller, 2021-12-26:
left debian experimental, right Fedora 35

```diff
# diff codec2/freedv_api.h /usr/include/codec2/freedv_api.h 
12,13c12
<   3. Simple demo programs in the "demo" directory
<   4. The full featured command line freedv_tx.c and freedv_rx.c programs
---
>   3. The sample freedv_tx.c and freedv_rx.c programs
148a148
> typedef void (*freedv_callback_rx_sym)(void *, _Complex float, float);
214a215
> void freedv_set_callback_txt_sym        (struct freedv *freedv, freedv_callback_rx_sym rx, void *callback_state);
```

So debian just straight up removed that callback type from codec2

marcusmueller, 2021-12-26:
yeah... `float _Complex` is C99, [`_Complex float`](http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1256.pdf) isn't. This is > 50% codec2's fault. So, Mait's solution is not totally absurd. It would have sufficed to 

```C
typedef void (*freedv_callback_rx_sym)(void *, float _Complex, float);
```

though.

How does that help us? Don't

mormj, 2022-01-20:
Closing since #5416 was merged

closed 2022-01-20 by mormj