gnuradio/gnuradio #5414: gr-iio: Pluto source block depends on existing samp_rate variable

aferragu, 2021-12-24:
### What happened?

When using the Pluto Source block in gnuradio-companion, any flowgraph won't work if the global variable samp_rate is not defined. I usually change samp_rate to RF_samp_rate to indicate the RF sampling rate and use other variants (e.g. audio_samp_rate, bb_samp_rate...) 

The problem seems to be this line:
https://github.com/gnuradio/gnuradio/blob/28b563c77233ea900b6c5f938d7aeb6fd07e38f1/gr-iio/grc/iio_pluto_source.block.yml#L28

which defines the default value of the sample rate in the block. If you remove the global samp_rate, then:
https://github.com/gnuradio/gnuradio/blob/28b563c77233ea900b6c5f938d7aeb6fd07e38f1/gr-iio/grc/iio_pluto_source.block.yml#L94
https://github.com/gnuradio/gnuradio/blob/28b563c77233ea900b6c5f938d7aeb6fd07e38f1/gr-iio/grc/iio_pluto_source.block.yml#L101

also fail since they depend on the previously defined value.

### System Information

OS: Fedora Workstation 35 (Linux)
GR Installation Method: conda install -c conda-forge/label/gnuradio_rc

### GNU Radio Version

3.10-git (master)

### Specific Version

3.10.0.0-rc2

### Steps to Reproduce the Problem

Create a simple flowgraph with one Pluto SDR Source and a QT Sink (e.g frequency sink).

### Relevant log output

```shell
Param - Fpass (Hz)(fpass):
	Value "samp_rate/4" cannot be evaluated:
	name 'samp_rate' is not defined

Param - Fpass (Hz)(fpass):
	Expression None is invalid for type 'float'.

Param - Fstop (Hz)(fstop):
	Value "samp_rate/3" cannot be evaluated:
	name 'samp_rate' is not defined

Param - Fstop (Hz)(fstop):
	Expression None is invalid for type 'float'.
```

marcusmueller, 2021-12-24:
But that's a very common pattern – the Signal Source block does the same, so do the UHD Source source and many others.

I'd say using a rather canonical variable name there is the most sane thing to do, right?

aferragu, 2021-12-27:
I don't get your answer. I can use a Qt Frequency sink with any variable name I like. In fact, when working with multiple sample rates in the same flowgraph I think it's a pretty bad idea to have a variable named samp_rate.

What I was reporting is that the Pluto source won't even run the flowgraph unless samp_rate is defined somewhere. I think hard coded variable names are not good.

marcusmueller, 2021-12-27:
> What I was reporting is that the Pluto source won't even run the flowgraph unless samp_rate is defined somewhere. 

Hm, maybe I'm misunderstanding things here:

I thought I wasn't able to reproduce a *problem*; just that the block turns red, the "Sample Rate" field, too, and generation of the flowgraph didn't work. 

But: now I understand what you mean, and I need to apologize for stopping short of a full reproduction:

[image]

The point you're raising is that there's no way to *fix* this, and that for some reason a global variable is used under the hood. Yeah, this is a severe bug.

aferragu, 2021-12-27:
Thanks! Re-reading my post I noticed that the "way to reproduce the problem" wasn't clear at all. Global variable under the hood is exactly what I meant.

This wasn't present in GNURadio 3.8 (I've just recently switched to 3.10 to take advantage of gr-iio being in tree).

marcusmueller, 2021-12-27:
@aferragu I've been playing around with this last night, and the way GRC fills in these defaults makes it impossible to use anything "generated" (like in the `hide:` expression directly below). So, setting all these to 0. People selecting "design" as filter method is a clear indication of them willing to figure out what they want for these parameters – no need to use any defaults.

marcusmueller, 2021-12-27:
@aferragu would you mind testing whether https://github.com/gnuradio/gnuradio/pull/5420 works for you? I don't have hardware to verify!

aferragu, 2021-12-27:
@marcusmueller I'm still a bit new to the GNURadio build system. In fact I'm using the conda installation with the release candidate because I always end up in trouble compiling from source.
But if you point me in the right direction on how to compile #5420 to test it I'll be glad to do it.

marcusmueller, 2021-12-27:
@aferragu  Ha! Your problem report is so well-detailed, I forgot that this might be non-trivial to you!

Good news: no need to compile at all, just click on the "files" tab in #5420, then scroll to `iio_pluto_source.block.yml`, click on the three dots ···, view File, "Raw", download that and replace the file of the same name in your installation with it :) Then, re-open GRC, add a Pluto Source, delete the samp_rate variable, set up the block correctly, connect it to a Null sink (or anything else) and just see if the flow graph builds & runs. (I'm here stuck in cascading issues from trying to make a remote IIO device work... ugh, more problems)

If this is more effort than you happen to have time for, it's OK, just let me know :)

aferragu, 2021-12-27:
Ok I downloaded the .ymls and replaced the original ones in two places with conda, but seems gnuradio-companion is not using them? I still get the same error and the hidden defaults still show "samp_rate".
I don't like conda and seems to me that testing there is not a good idea :-/ 
Maybe I just put them in the wrong place....

marcusmueller, 2021-12-27:
ah, Josh went ahead and merged these fixes, so I guess they were passing his code review :) for now that means this is (probably) fixed upstream

aferragu, 2021-12-27:
Closed by #5420. Will repoen if more testing required.

closed 2021-12-27 by aferragu