gnuradio/gnuradio #5415: grc: make it clearer why the project cannot be executed

d33tah, 2021-12-26:
### Feature Description

Thanks for creating gnuradio! I recently became interested in signal processing and while I felt that I became acquainted with the basics, I decided to give Gnuradio Companion a try once again. Previously I found the UX overwhelming and this time I decided to spend more time trying to understand it. In doing that, I understood one of the problems I might have had. Basically, I tried to create a "hello, world" project. I started GRC, added random source, audio sink, wired them together, then tried to run the prototype... and found the "execute" button grayed out.

My first assumption was that the program was for some reason broken on my system and started Googling, but it's not easy to Google for UX problems. There weren't many hits for "gnuradio companion execute grayed out" or "disabled". It took some experimenting with the GUI to find view->flowgraph errors and then some more to understand the cryptic "ID "default" is blacklisted" error message (which was there because there's no project name set by default). I think that it's an easy problem to solve, here's my suggestion:

Instead of graying "execute" out with no feedback, either add a hint on the button and menu item (something saying "problems found - check view->flowgraph errors for suggestions") or straight up enable the user to press execute and then display the actual list of errors, with a red caption explaining why execution failed.

What do you think of it? I think that it's not necessarily difficult to implement and would simplify things for newcomers. I for one would have appreciate it.

### Feature Urgency

high (impacts broad portion of ecosystem)

### More Information

I marked this issue as high urgency because I believe that lack of this feature significantly increases barrier to entry for new users.

777arc, 2021-12-26:
Does the default ID have to be blacklisted?  It wasn't always, and there is something to say for the convenience of being able to quickly open a new flowgraph and test something without setting the ID.  What are we trying to protect by blacklisting it?  Is it just to help enforce good practice?  Also what if we let the ID be empty and then if it's empty we just use the name of the grc file?

mormj, 2022-01-20:
This has been addressed with #5421 - the flowgraph no longer starts in an "errored" state, and the id is set to whatever the flowgraph is saved as if it is left as "default".

closed 2022-01-20 by mormj