gnuradio/gnuradio #5418: GRC: Add "Open Examples" menu item

marcusmueller, 2021-12-26:
### Feature Description

As raised in #5417, the examples that come with GNU Radio installations are hard to find.

Adding a menu item to the file menu "Open examples" might remedy that.

### Feature Urgency

low (just an idea)

### More Information

It's really not clear to me whether that is a *good* idea. 

* (+) It's a shame we ship examples, but only developers are prone to know about them
* (-) Cluttering the File menu is an absolute antipattern
* (+) It's an operation on a file
* (-) #5417 suggests that the user's expectation would be that the examples are a good starting point. Which they IMHO aren't – an integrated tutorial would be. Fostering that expectation would be didactically misleading for all beginners but these that come from an SDR background already, which are the users with the flattest learning curve, probably.
  * (+) in absence of the tutorials, wouldn't access to the examples still be better?

Hence, looking for guidance, especially from @ei8fdb, @duggabe and @marc

ei8fdb, 2022-01-26:
@marcusmueller I've tried to add the UX label to this issue but can't seem to. Can you add it please? I'm doing some design ideas at the moment on how this might work.

marcusmueller, 2022-01-26:
@ei8fdb  done