gnuradio/gnuradio #5429: GRC3.9.3.0 MAC OSX 10.15 + xquartz | Scaling and garbled display issues

m-anish, 2021-12-28:
### What happened?

Installed  gnuradio-companion on mac osx via homebrew. When it launches, and I make a simple flow graph the window resolution inside gnuradio-companion doubles or so and nothing is clickable anymore.

My installation process was as follows:
1. Installed xquartz
2. Installed homebrew
3. Installed soapy libraries via homebrew
4. Installed gnuradio via homebrew

I know that there were issues in the past pertaining to screen resolution but they are all closed, and I am looking for a simple fix (if possible). Thanks!

[image]

### System Information

OS: Mac OSX 10.15.7 Catalina
GR Installation Method: homebrew

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

3.9.3.0

### Steps to Reproduce the Problem

* Launch gnuradio-companion from the terminal. 
* Create a simple flow graph and save it
* Somewhere during this process, the flow window's icons and graphics all double in size and make the interface unusable.

citizenrich, 2021-12-28:
@m-anish Have you had a try with this fix? https://github.com/gnuradio/gnuradio/pull/4814#issuecomment-925190019 IIRC that's what worked for me with macOS/Homebrew

m-anish, 2021-12-28:
I was just looking at that and the eventual commit that made it to 3.9.4.0

https://github.com/gnuradio/gnuradio/commit/6cdcc2e432830d150274026a6033f40339a7d5a7

I am on 3.9.3.0 and this is what homebrew provides, but I'll try to find a way to get it into my setup.

Thanks!

m-anish, 2021-12-28:
@citizenrich I'm new to homebrew, and generally new to gnuradio in terms of hacking on its code. Is it as simple as making the edit in a python script, or do I need to edit the homebrew formula, apply a patch, and recompile?

m-anish, 2021-12-28:
FWIW, I just made the change in the following file
'/usr/local/lib/python3.9/site-packages/gnuradio/grc/gui/DrawingArea.py'

and it _seems_ to do the job.

I'll keep this bug report open a little while to see if there are any other issues, but so far so good!

citizenrich, 2021-12-28:
Yep, you beat me to it. Exactly that. However, when a new version of GR comes out I think the edit will be blown away, so you'll have to do it again. Hope that helps!

citizenrich, 2021-12-28:
Also, check out the examples at `/usr/local/Cellar/gnuradio/3.9.3.0_1/share/gnuradio/examples` and you want to run `volk_profile`.

m-anish, 2021-12-29:
> Also, check out the examples at `/usr/local/Cellar/gnuradio/3.9.3.0_1/share/gnuradio/examples` and you want to run `volk_profile`.

Thanks. I'll look at the examples and I did run volk_profile.

I think the fix was cherrypicked onto 3.9.4.0 which would be the version next to the version I'm on 3.9.3.0, so when it does update, I shouldn't ideally have to do anything.

Thanks for your help!

m-anish, 2021-12-29:
Uh oh! There is still an issue.

When I open a new tab, the display again goes haywire, and I see these messages...

```
(gnuradio-companion:90023): Gtk-WARNING **: 10:15:51.299: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node header owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:90023): Gtk-WARNING **: 10:15:51.299: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tabs owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:90023): Gtk-WARNING **: 10:15:51.299: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tab owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:90023): Gtk-WARNING **: 10:15:51.299: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tab owner gnuradio+grc+gui+Notebook+Notebook)
```

m-anish, 2021-12-29:
Hmm, more notes. This issue happens when in Display settings, I turn scaling off. When scaling is on, the issue does not seem to happen.

I would still characterize this as a bug even though there is a workaround for it.

citizenrich, 2021-12-29:
I also see this issue. I'll try your fix. Where did you turn display off?

m-anish, 2021-12-29:
It is in system settings -> display.

citizenrich, 2021-12-29:
Ok thanks. I thought you meant another X or similar display parameter. Doh.

marcusmueller, 2024-01-31:
I think this is another symptom of what #2726 addresses; I'm pretty confident that if this was still happening under 3.10, we'd hear a lot about it; so: closing it!

If this is still applicable and you're running the latest 3.9 or 3.10, please reopen.

closed 2024-01-31 by marcusmueller