gnuradio/gnuradio #5430: Issues in polar code module

dfdtc, 2021-12-29:
After some test of polar code module (mainly using the polar_code_example.grc),I found some problem, maybe they are bugs or not.  
1. I can't set the channel type to AWGN, or it will give me a error:(only show a part of tracking message here)
File "/usr/lib/python3/dist-packages/gnuradio/fec/polar/channel_construction_awgn.py", line 197, in upper_convolve
    q[0, idx] = (tpm[0 / i] ** 2 + tpm[1, i] ** 2, 2)
ZeroDivisionError: division by zero
2. When I testing the encoder and decoder(SC and SCL)(using the original setting in the example(polar_code_example.grc)), I found that I can't decoding the message correctly, even the amplitude of noise have set to 0(meant no noise in transmitting). BUT, if I replace the encoding(and decoding) module from POLAR to LDPC, this example can give the correct output(by viewing the GUI output of Coded window).
I'm not a researcher in this field, I just want to use the module to finish my own project. So.....sorry,I just know I have some problem. But I can't give a suggestion or some worked solutions for them.
Thank you for your notice of this issue.

marcusmueller, 2021-12-29:
Hi! Could you add the full error, and also maybe a picture / writedown of the exact parameters used? 

This does sound like a bug, so thanks for reporting!

willcode, 2021-12-29:
`upper_convolve()` in `channel_construction_awgn()` looks like it will always fail with a `/0` since `i in range(mu)`.

dfdtc, 2021-12-30:
(Working flow and parameters)
In conclusion:those flow are based on the example in "[/gnuradio/gr-fec/examples/polar_code_example.grc](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/examples/polar_code_example.grc)". Only removed the part of uncoded signal displaying.
**Polar_code testing flow:**
[image]
**LDPC_code testing flow:**
[image]
**Polar_code encoding and decoding(SC and SCL)(with noise)**
[image]
**Polar_code encoding and decoding(SC and SCL)(WITHOUT noise)**
[image]
**LDPC encode and decode(with low amplitude of noise)**
[image]
[image]
**LDPC encode and decode(with high amplitude of noise)**
[image]
_**The message in testing.txt (the "file source" block in working flow)is the repeating of text "testing"**_
**(Just in my point of view)**
As the picture showed, the polar_code module cannot decode the coding produced by polar_code_encoder(both SC and SCL), in both of noisy and noiseless environments. BUT, just replaced the encoder and decoder from POLAR to LDPC, this flow can give the right output(corrected message in low noise environment, partly corrected message in high noise environment).

VladislavF, 2023-01-06:
I can confirm the same issue occurs in GR 3.10.4

VladislavF, 2023-01-06:
> `upper_convolve()` in `channel_construction_awgn()` looks like it will always fail with a `/0` since `i in range(mu)`.

Is there any news on resolving this? Was this a syntax error or does it go deeper? I would offer my help, but I'm afraid I know absolutely nothing about polar codes, so I don't have any intuition here.

willcode, 2023-01-06:
There haven't been any changes to this file and `upper_convolve()` still looks really messed up. `tpm[0 / i]` is both useless and crashy. I don't know anything about polor codes either. @jdemel ?

jdemel, 2023-01-07:
A lot of papers on polar codes don't use this method for construction. Most resort to the Bhattacharyya Bounds method (the BEC channel option), or Gaussian Approximation. Also, Beta-Expansion is an interesting candidate that might produce the best results in the end.

The whole "channel" argument refers to the channel construction algorithm and its underlying assumption. You can use any and the results will mostly be the same. They vary be less than .1 dB and the preference for one algorithm over another may change if you choose a different polar decoder.
In retrospect it would've been better to call this argument "algorithm" and options would be: BB, GA, BE, DE.

The example flowgraph in its default configuration shouldn't break though.

PR #6457 (2023-01-09): fec: fix typo in upper_convolve()

## Description
Type in `upper_convolve` was generating /0 error, and doesn't look like the intended code. Note that I don't know anything about polar codes!

## Related Issue
Fixes #5430 

## Testing Done
None

## Checklist
- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md#dco-signed)
- [x] My code follows t [... cut]
closed 2023-01-09 by willcode