gnuradio/gnuradio #5431: Gnuradio-companion tends to crash on macOS 10.15.7 when connecting blocks

rl1987, 2021-12-29:
### What happened?

A bug happened!

### System Information

OS: macOS 10.15.7
GR Installation Method: Homebrew

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

3.9.3.0

### Steps to Reproduce the Problem

Launch gnuradio-companion and any two blocks. Try connecting the output of the first block to the input of the second one. This tends to cause a crash.

Note that this started happening not instantly, but after I was playing with gnuradio-companion for a bit.

### Relevant log output

```shell
2021-12-29 14:52:12.650 Python[75255:1441851] unlockFocus called too many times. Called on <GdkQuartzView: 0x7f8a2a133c80>.
Segmentation fault: 11
```
```

rl1987, 2021-12-29:
[crash_reports.zip](https://github.com/gnuradio/gnuradio/files/7788974/crash_reports.zip)

DragosRotaru, 2022-03-09:
OS: macOS 10.15.7
GR Installation Method: Homebrew
GNU Radio Version: 3.9 (maint-3.9)
GNU Radio Companion Version: 3.9.3.0

I've reproduced the same error, here is my log output:

`2022-03-09 10:05:23.199 Python[76052:1650503] unlockFocus called too many times. Called on <GdkQuartzView: 0x7fcdecf59380>.
Segmentation fault: 11`

The GUI also zooms in when this happens and the arrow is not centered on the mouse anymore

ttomasini, 2022-05-24:
Similar issue here
OS: macOS 11.6.6
GR Installation Method: Homebrew
GNU Radio Version: 3.9 (maint-3.9)
GNU Radio Companion Version: 3.9.3.0

Steps to reproduce:
1. Add compatible blocks (I added a noise source and a low pass filter)
2. Connect *out* of the first block with the *in* of the second block with drag and drop
3. Enjoy the segfault below

```sh
❯ gnuradio-companion --log debug
[DEBUG] Starting GNU Radio Companion (3.9.13) (main.py:67)
[DEBUG] Loading platform (main.py:73)
Slow YAML loading (libyaml not available)
[DEBUG] Loading application (main.py:82)
[DEBUG] Running (main.py:84)
<<< Welcome to GNU Radio Companion 3.9.3.0 >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks

Loading: "/Users/tio/test.grc"
>>> Done
Python(37013,0x115207e00) malloc: can't allocate region
:*** mach_vm_map(size=1125870791806976, flags: 100) failed (error code=3)
Python(37013,0x115207e00) malloc: *** set a breakpoint in malloc_error_break to debug
[1]    37013 segmentation fault  gnuradio-companion --log debug
```

ryanvolz, 2022-11-10:
Any chance https://github.com/gnuradio/gnuradio/pull/6352 fixes this? That's a python-only fix, so you can manually edit the appropriate installed python files to test it.

closed 2022-11-17 by mormj