gnuradio/gnuradio #5435: RuntimeError: the sip module implements API v12.0 to v12.8 but the PyQt5.QtCore module requires API v12.9

nminpc, 2021-12-30:
### What happened?

I can't find a solution to the problem.

File "/usr/lib/python3/dist-packages/gnuradio/qtgui/__init__.py", line 38, in <module>
    from .range import Range, RangeWidget
  File "/usr/lib/python3/dist-packages/gnuradio/qtgui/range.py", line 24, in <module>
    from PyQt5 import Qt, QtCore, QtWidgets
RuntimeError: the sip module implements API v12.0 to v12.8 but the PyQt5.QtCore module requires API v12.9

### System Information

OS: (Linux)
(gr-gsm pybombs)

### GNU Radio Version

3.8 (maint-3.8)

### Specific Version

3.8.2.0

### Steps to Reproduce the Problem

File "/usr/lib/python3/dist-packages/gnuradio/qtgui/__init__.py", line 38, in <module>
    from .range import Range, RangeWidget
  File "/usr/lib/python3/dist-packages/gnuradio/qtgui/range.py", line 24, in <module>
    from PyQt5 import Qt, QtCore, QtWidgets
RuntimeError: the sip module implements API v12.0 to v12.8 but the PyQt5.QtCore module requires API v12.9

willcode, 2022-01-03:
@nminpc Which Linux distro and version?

Porcupine1, 2022-01-04:
I was facing same problem not long ago: but on windows. I fixed it by upgrading sip and PyQt5-sip.

_1. pip install sip --upgrade
2. pip install PyQt5-sip-sip --upgrade_

totomobile43, 2022-01-26:
> pip install PyQt5-sip-sip --upgrade

Worked for me as well (and I am on macos), but its just one -sip: `pip install PyQt5-sip--upgrade`

marcusmueller, 2022-01-27:
how did @Porcupine1 and @totomobile43  install GNU Radio? That you can upgrade sip and it fixes things indicates the installation method did not correctly depend on the right version.

Varat7v2, 2022-06-04:
> > pip install PyQt5-sip-sip --upgrade
> 
> Worked for me as well (and I am on macos), but its just one -sip: `pip install PyQt5-sip --upgrade`

Thanks this solved my issue.

closed 2022-06-14 by willcode

yahiamoussadek, 2022-11-02:
i used the command line that solved the issue for almost everyone, except for me