gnuradio/gnuradio #5436: hier_block2::lock() crash/freeze after disconnecting at one block from it's output

vladisslav2011, 2021-12-30:
### What happened?

hier_block2::lock() freezes/crashes with segfault if called after disconnecting a block from heir_block2 output port.

### System Information

OS: Linux
GR Installation Method: native distro packages, Source

### GNU Radio Version

3.10-git (master)

### Specific Version

3.10.0.0-rc2

### Steps to Reproduce the Problem

1. Compile test program `g++ main.cpp -o main -lboost_system -lfmt -lgnuradio-runtime -lgnuradio-pmt -lvolk -lgnuradio-blocks 
`
2. Run the program `./main`
main.cpp:
```
#include <iostream>

#include <gnuradio/blocks/null_sink.h>
#include <gnuradio/blocks/null_source.h>
#include <gnuradio/hier_block2.h>
#include <gnuradio/blocks/copy.h>
#include <gnuradio/top_block.h>

class test_block : public gr::hier_block2
{

public:
typedef std::shared_ptr<test_block> sptr;
    static test_block::sptr make()
    {
        return gnuradio::get_initial_sptr(new test_block());
    }
    test_block();
    virtual ~test_block();
    void trigger();

private:
    gr::blocks::copy::sptr d_copy;
};

test_block::test_block(): gr::hier_block2 ("test_block",
                          gr::io_signature::make (1, 1, sizeof(gr_complex)),
                          gr::io_signature::make (1, 1, sizeof(gr_complex)))

{
    d_copy = gr::blocks::copy::make(sizeof(gr_complex));
    connect(self() , 0 , d_copy , 0);
    connect(d_copy , 0, self() , 0);
}

test_block::~test_block()
{
}

void test_block::trigger()
{
    std::cerr<<"test_block->trigger"<<std::endl;
    this->lock();
    std::cerr<<"test_block->lock"<<std::endl;
    this->unlock();
    std::cerr<<"test_block->unlock"<<std::endl;
}

int main()
{
    auto tb = gr::make_top_block("buggy");
    auto test0 = test_block::make();
    auto src0 = gr::blocks::null_source::make(sizeof(gr_complex));
    auto snk0 = gr::blocks::null_sink::make(sizeof(gr_complex));
    auto snk1 = gr::blocks::null_sink::make(sizeof(gr_complex));
    tb->lock();
    std::cerr<<"tb->lock"<<std::endl;
    tb->connect(src0, 0, test0, 0);
    tb->connect(test0, 0, snk0, 0);
    tb->connect(test0, 0, snk1, 0);
    tb->unlock();
    std::cerr<<"tb->unlock"<<std::endl;
    tb->lock();
    std::cerr<<"tb->lock"<<std::endl;
    tb->disconnect(test0, 0, snk1, 0);
    tb->unlock();
    std::cerr<<"tb->unlock"<<std::endl;
    test0->trigger();
    std::cerr<<"TEST PASSED!!!"<<std::endl;
    return 0;
}
```

### Relevant log output

```shell
tb->lock
tb->unlock
tb->lock
tb->unlock
test_block->trigger
Segmentation fault (core dumped)
```

vladisslav2011, 2021-12-30:
Backtrace:
```
Thread 1 "main" received signal SIGSEGV, Segmentation fault.
0x00007ffff791b673 in __cxxabiv1::__vmi_class_type_info::__do_dyncast(long, __cxxabiv1::__class_type_info::__sub_kind, __cxxabiv1::__class_type_info const*, void const*, __cxxabiv1::__class_type_info const*, void const*, __cxxabiv1::__class_type_info::__dyncast_result&) const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff791b673 in __cxxabiv1::__vmi_class_type_info::__do_dyncast(long, __cxxabiv1::__class_type_info::__sub_kind, __cxxabiv1::__class_type_info const*, void const*, __cxxabiv1::__class_type_info const*, void const*, __cxxabiv1::__class_type_info::__dyncast_result&) const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff7918b23 in __dynamic_cast () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007ffff798d75f in std::ctype<char> const& std::use_facet<std::ctype<char> >(std::locale const&) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff7e3277a in boost::basic_format<char, std::char_traits<char>, std::allocator<char> >::basic_format(char const*) () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#4  0x00007ffff7e8db98 in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#5  0x00007ffff7e8df0a in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#6  0x00007ffff7e8df0a in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#7  0x00007ffff7e8df0a in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#8  0x00007ffff7e8df0a in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
(...)
#10697 0x00007ffff7e8df0a in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#10698 0x00007ffff7e8df0a in gr::hier_block2_detail::lock() () from /usr/local/lib/libgnuradio-runtime.so.3.10.0
#10699 0x0000555555572bfe in test_block::trigger() ()
#10700 0x0000555555573058 in main ()
```

argilo, 2021-12-30:
The following Python code can also reproduce the issue. In 3.10 it produces a segfault (due to a stack overflow) and in 3.9 & 3.8 it produces an infinite loop (100% CPU) utilization.
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

#
# SPDX-License-Identifier: GPL-3.0
#
# GNU Radio Python Flow Graph
# Title: Crash Test
# Author: Clayton Smith
# GNU Radio version: 3.10.0.0-rc2

from gnuradio import blocks
from gnuradio import gr

class hier_test(gr.hier_block2):
    def __init__(self):
        gr.hier_block2.__init__(
            self, "Hierarchical Block Test",
            gr.io_signature(1, 1, gr.sizeof_gr_complex*1),
            gr.io_signature(1, 1, gr.sizeof_gr_complex*1),
        )

        self.blocks_copy_0 = blocks.copy(gr.sizeof_gr_complex*1)

        self.connect((self.blocks_copy_0, 0), (self, 0))
        self.connect((self, 0), (self.blocks_copy_0, 0))

class lock_bug_test(gr.top_block):
    def __init__(self):
        gr.top_block.__init__(self, "Bug Test")

        self.hier_test_0 = hier_test()
        self.blocks_vector_source_x_0 = blocks.vector_source_c((0, 0, 0), False, 1, [])
        self.blocks_null_sink_0 = blocks.null_sink(gr.sizeof_gr_complex*1)
        self.blocks_null_sink_1 = blocks.null_sink(gr.sizeof_gr_complex*1)

        self.connect((self.blocks_vector_source_x_0, 0), (self.hier_test_0, 0))
        self.connect((self.hier_test_0, 0), (self.blocks_null_sink_0, 0))
        self.connect((self.hier_test_0, 0), (self.blocks_null_sink_1, 0))

    def trigger_bug(self):
        self.disconnect((self.hier_test_0, 0), (self.blocks_null_sink_1, 0))
        self.hier_test_0.lock()

def main(top_block_cls=lock_bug_test, options=None):
    tb = top_block_cls()
    tb.trigger_bug()

if __name__ == '__main__':
    main()
```

argilo, 2021-12-30:
The bug is very likely present in 3.7 as well, since Gqrx appears to have implemented a workaround for it back in 2014: https://github.com/gqrx-sdr/gqrx/commit/eca0ea7163ff16abd698737fc8ac1baab21be668

argilo, 2021-12-30:
I would guess the bug lies in `hier_block2_detail::disconnect`'s tracking of `d_parent_detail`.

vladisslav2011, 2021-12-30:
Yes. That's incorrect tracking of `d_parent_detail`.
`d_parent_detail` should be cleared only when all ports (and message ports?) get disconnected, not the first one.

argilo, 2021-12-30:
Even just instantiating a hierarchical block and immediately calling `lock()` is enough to reproduce a crash:
```python
from gnuradio import blocks
from gnuradio import gr

class hier_test(gr.hier_block2):
    def __init__(self):
        gr.hier_block2.__init__(
            self, "Hierarchical Block Test",
            gr.io_signature(1, 1, gr.sizeof_gr_complex*1),
            gr.io_signature(1, 1, gr.sizeof_gr_complex*1),
        )

        self.blocks_copy_0 = blocks.copy(gr.sizeof_gr_complex*1)

        self.connect((self.blocks_copy_0, 0), (self, 0))
        self.connect((self, 0), (self.blocks_copy_0, 0))

ht = hier_test()
ht.lock()
```

argilo, 2021-12-30:
I had a closer look here, and it seems the way things are supposed to work is that when `lock()` or `unlock()` is called on a hierarchical block, the request is forwarded up the hierarchy to the top block. The `d_parent_detail` member variable stores a link to the parent of the hierarchical block, if it exists.

There seems to be two bugs here:

1. When a hierarchical block has no parent, `hier_block2_detail::lock()` and `hier_block2_detail::unlock()` fall back to calling `d_owner->lock()` / `d_owner->unlock()`, which immediately results in infinite recursion because `hier_block2::lock()` and `hier_block2::unlock()` just forward the request straight back to `hier_block2_detail`. In this situation, it would probably be better to raise an exception.
2. `hier_block2_detail`'s `disconnect` and `msg_disconnect` methods clear `d_parent_detail` as soon as the first external block is disconnected. It should only do that when the last external block is disconnected.

vladisslav2011, 2021-12-30:
> Even just instantiating a hierarchical block and immediately calling `lock()`

That's different issue, that comes from incorrect checking for `top_block`. It is assumed, that if `d_parent_detail == 0`, then `d_owner` points to `top_block`, but this is true when the `hier_block2` is just created or after it's port disconnection. I think, we should check for `top_block` with `dynamic_cast<top_block *>(d_owner)`...

vladisslav2011, 2021-12-30:
> 2\. It should only do that when the last external block is disconnected.

I've implemented simple counting to track usage (see https://github.com/vladisslav2011/gnuradio/tree/fix_5436), but that's not enough. There is `hier_block2::disconnect_all()`, that does not call `disconnect`/`msg_disconnect` on  individual block, but just clears the flowgraph. It looks, like we should go hard way and check all edge and block lists for our block presence on each `connect` call...

argilo, 2021-12-30:
> It is assumed, that if `d_parent_detail == 0`, then `d_owner` points to `top_block`

Right. I see the same assumption is made in `hier_block2_detail::flatten_aux`.

argilo, 2021-12-30:
> There is `hier_block2::disconnect_all()`, that does not call `disconnect`/`msg_disconnect` on individual block, but just clears the flowgraph.

Doesn't it clear just the internal connections within the hierarchical block?

argilo, 2021-12-30:
If that's the case, then `disconnect_all()` wouldn't change the external connections or reference count.

vladisslav2011, 2021-12-30:
Yes. it clears internal connections, but what would happen when there is another hier_block2 inside current hier_block2?

argilo, 2021-12-30:
Ah, I guess that might be a problem.

closed 2022-01-21 by mormj