gnuradio/gnuradio #5437: "Hilbert" block doesn't calculate (only) the Hilber transform, but the analytic signal

marcusmueller, 2021-12-31:
### What happened?

The API documentation says

> hilbert_fc
> Hilbert transformer.
> real output is input appropriately delayed. imaginary output is hilbert filtered (90 degree phase shift) version of input.
> Constructor Specific Documentation:
> Build a Hilbert transformer filter block.
> Args:
>     ntaps : The number of taps for the filter.
>     window : Window type (see `fft::window::win_type`) to use.
>     param : Parameter value for a Kaiser/Exp/Gaussian/Tukey window.

Now, what input + H(input) for a real input is, it's the analytic signal.

The bug here is merely naming; also, it would seem to be nice to have "Hilbert transformer taps" as separate block, just as we have "low pass filter taps", etc, especially since chances are you'd want your discrete HT to be of significant length, i.e. use an FFT filter.

### System Information

--

### GNU Radio Version

3.10-git (master)

### Specific Version

since the beginning of time (at least 3.6.2)

### Steps to Reproduce the Problem

--

valdaarhun, 2022-02-10:
Hi. I am new to gnuradio and I would like to work on this issue. I have read through the [PySDR](https://pysdr.org/) tutorial so I believe I have been acquainted with signals and systems, and dsp in general.

valdaarhun, 2022-02-15:
Hi. I wanted to confirm the changes that are to be done in this issue.

Should the name `hilbert_fc` be changed to something else, for eg: `analytic_signal_fc`? And then should a new `hilbert_fc` block be created that calculates `H(input)`?

marcusmueller, 2022-06-26:
Hi @valdaarhun, sorry for the long silence.

> Should the name hilbert_fc be changed to something else, for eg: analytic_signal_fc? 

No! That's far too intrusive, as it breaks existing code, only to have better naming. All that needs to change is the name in the .grc Block definition file.

> And then should a new hilbert transformer tap be created that calculates H(input)?

no.

valdaarhun, 2022-09-17:
Hi. I am sorry for the really late response. I think I have understood what I need to do here. I believe I need to change the description given in the `hilbert_fc.h` file.

valdaarhun, 2022-09-17:
Hi. I have opened a PR in #6177. Please do let me know if there are changes that need to be incorporated.

> it would seem to be nice to have "Hilbert transformer taps" as separate block, just as we have "low pass filter taps", etc, especially since chances are you'd want your discrete HT to be of significant length, i.e. use an FFT filter.

Sorry, I didn't exactly understand this. I see the hilbert transformer block already exists. On the other hand, I don't see any existing block for generating the analytic signal.

closed 2022-10-24 by mormj