gnuradio/gnuradio #5445: SegFault when adding rtlsdr flow

johnavich, 2022-01-05:
### What happened?

When adding a GRC flow, the app crashes with the following in syslog:
Jan  4 19:50:21 np3 kernel: [79027.967650] gnuradio-compan[235862]: segfault at 7f1642acbab7 ip 00007f15e6bc58ea sp 00007ffdefdd5170 error 7 in _imagingft.cpython-38-x86_64-linux-gnu.so[7f15e6bbd000+b000]
Jan  4 19:50:21 np3 kernel: [79027.967658] Code: e8 7b ae ff ff 48 85 c0 48 89 03 0f 84 bf 01 00 00 48 8d 35 28 11 00 00 48 89 c7 e8 10 b3 ff ff 48 8b 2d 29 1f 20 00 48 85 c0 <48> 89 45 00 0f 84 6c 01 00 00 48 8b 3b 48 8d 35 9a 10 00 00 e8 ed

### System Information

OS: Ubuntu 20.04.3
GR Installation Method: apt

### GNU Radio Version

3.8 (maint-3.8)

### Specific Version

3.8.3.1

### Steps to Reproduce the Problem

Add flow for any radio from gr-smart_meters.  Segfault occurs immediately.

### Relevant log output

```shell
Jan  4 19:50:21 np3 kernel: [79027.967650] gnuradio-compan[235862]: segfault at 7f1642acbab7 ip 00007f15e6bc58ea sp 00007ffdefdd5170 error 7 in _imagingft.cpython-38-x86_64-linux-gnu.so[7f15e6bbd000+b000]
Jan  4 19:50:21 np3 kernel: [79027.967658] Code: e8 7b ae ff ff 48 85 c0 48 89 03 0f 84 bf 01 00 00 48 8d 35 28 11 00 00 48 89 c7 e8 10 b3 ff ff 48 8b 2d 29 1f 20 00 48 85 c0 <48> 89 45 00 0f 84 6c 01 00 00 48 8b 3b 48 8d 35 9a 10 00 00 e8 ed
```

willcode, 2022-01-05:
@johnavich Does this happen with any built-in GNU Radio blocks? The gr-smart_meters package is not supported by the main GNU Radio project so you would submit the issue to https://github.com/BitBangingBytes/gr-smart_meters/issues.

marcusmueller, 2022-01-05:
Since the segfault happens in a library that's not used by GNU Radio itself, 

```
_imagingft.cpython-38-x86_64-linux-gnu.so
```

I'm afraid that's nothing GNU Radio as a project itself can help you sort out; so, closing it on this issue tracker.

closed 2022-01-05 by marcusmueller