gnuradio/gnuradio #5447: Problem with implementing tests

nedsana, 2022-01-07:
Hi, 
I recently created successfully my first custom C++ block, following the OutOfTreeModules tutorial and gr_modtool. 
I did it for GnuRadio v3.8.2.0-111-g6aad98a6 - compiled and installed from source.

Now I need to extend the implementation and I want to add new tests for particular block methods. And this is where I hit my problems.

Here is a simplified version of my implementation, just to illustrate the issue:

_**_File: gr-Dummy/include/Dummy/dummy.h:_**_
```
class Dummy_API Dummy : virtual public gr::sync_block
{
public:
    typedef boost::shared_ptr<Dummy> sptr;
    static sptr make();

    virtual void Dummy_Method(int& v) = 0;
};
```
_**File: gr-Dummy/lib/dummy_impl.h:**_
```
class Dummy_impl : public Dummy
{
public:
    Dummy_impl();
    ~Dummy_impl();

    void Dummy_Method(int& v);
};
```
_**File: gr-Dummy/lib/dummy_impl.cpp:**_
```
...
void Dummy_impl::Dummy_Method(int& v)
{
    v = 55;
}
...
```
_**File: gr-Dummy/python/qa_Dummy.py:**_
```
def test_020_t(self):
    out = 0;
    inst = Dummy.Dummy()
    inst.Dummy_Method(out)
    print("out={}".format(out))

```

Note that **"Dummy_Method"** takes a **_none-const reference argument_**, which will be used as in/out argument. 

All compiles fine, but when I execute the test I get:
```
Traceback (most recent call last):
  File "./gr-Dummy/python/qa_Dummy.py", line 585, in test_020_t
    inst.Dummy_Method(out)
  File "./gr-Dummy/build/swig/Dummy_swig.py", line 273, in Dummy_Method
    return _Dummy_swig.Dummy_sptr_Dummy_Method(self, v)
TypeError: in method 'Dummy_sptr_Dummy_Method', argument 2 of type 'int &'
```

If I understand it correct none-const reference arguments are not accepted by Swig. 
I briefly checked the documentation and it seems that Swig does allow none-const reference arguments. 
So it seems that my project needs to be reconfigured somehow, or none-const reference arguments are rejected on purpose. Unfortunately I failed to find any examples or discussions on the topic. 

Could you please comment on this and propose solution?

Another thing I tried was to create C++ tests. This, I think, would be sufficient for my purposes. But again I have problems. 
My test should make an instance of the Dummy_impl class like this:
```
#include <Dummy/Dummy.h>
#include "Dummy_impl.h"
#include <gnuradio/attributes.h>
#include <boost/test/unit_test.hpp>

namespace gr {
  namespace Dummy {

    BOOST_AUTO_TEST_CASE(test_Dummy_t1)
    {
        int out = 0;
        Dummy_impl inst;
        inst.Dummy_Method(out);
    }

  } /* namespace Dummy */
} /* namespace gr */
```
But then I have linker issues:
```
[  6%] Built target pygen_apps_9a6dd
[ 18%] Built target doxygen_target
[ 31%] Built target pygen_python_cac4b
[ 31%] Built target _Dummy_swig_doc_tag
[ 43%] Built target gnuradio-Dummy
Scanning dependencies of target Dummy_qa_Dummy.cc
[ 56%] Built target Dummy_swig_swig_doc
[ 62%] Built target Dummy_swig_swig_compilation
[ 75%] Built target Dummy_swig
[ 81%] Building CXX object lib/CMakeFiles/Dummy_qa_Dummy.cc.dir/qa_Dummy.cc.o
[ 93%] Built target pygen_swig_9fbf6
[100%] Linking CXX executable Dummy_qa_Dummy.cc
/usr/bin/ld: CMakeFiles/Dummy_qa_Dummy.cc.dir/qa_Dummy.cc.o: in function `gr::Dummy::test_Dummy_t1::test_method()':
qa_Dummy.cc:(.text+0x3d): undefined reference to `gr::Dummy::Dummy_impl::Dummy_impl()'
/usr/bin/ld: qa_Dummy.cc:(.text+0x4a): undefined reference to `gr::Dummy::Dummy_impl::Dummy_Method(int&)'
/usr/bin/ld: qa_Dummy.cc:(.text+0x52): undefined reference to `gr::Dummy::Dummy_impl::~Dummy_impl()'
/usr/bin/ld: qa_Dummy.cc:(.text+0x7f): undefined reference to `gr::Dummy::Dummy_impl::~Dummy_impl()'
/usr/bin/ld: CMakeFiles/Dummy_qa_Dummy.cc.dir/qa_Dummy.cc.o: in function `gr::Dummy::test_Dummy_t1_invoker()':
qa_Dummy.cc:(.text+0xea9): undefined reference to `gr::Dummy::Dummy_impl::Dummy_impl()'
/usr/bin/ld: qa_Dummy.cc:(.text+0xeb5): undefined reference to `gr::Dummy::Dummy_impl::Dummy_Method(int&)'
/usr/bin/ld: qa_Dummy.cc:(.text+0xebd): undefined reference to `gr::Dummy::Dummy_impl::~Dummy_impl()'
/usr/bin/ld: qa_Dummy.cc:(.text+0x2818): undefined reference to `gr::Dummy::Dummy_impl::~Dummy_impl()'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/CMakeFiles/Dummy_qa_Dummy.cc.dir/build.make:98: lib/Dummy_qa_Dummy.cc] Error 1
make[1]: *** [CMakeFiles/Makefile2:274: lib/CMakeFiles/Dummy_qa_Dummy.cc.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
This is seems like Cmake settings, but for now I have failed to find what should be changed and I am running out of ideas. 
Can you suggest anything?

Just in case I have attached the Dummy project.

Thank you in advance.

[gr-Dummy.zip](https://github.com/gnuradio/gnuradio/files/7827577/gr-Dummy.zip)

velichkov, 2022-01-07:
Hi @nedsana,

Welcome to GNU Radio!

Please use [chat.gnuradio.org](https://chat.gnuradio.org) or the [mailing list](https://wiki.gnuradio.org/index.php/MailingLists) for help requests. The issue tracker is used for bug reports and feature requests.

> My test should make an instance of the Dummy_impl class like this:

You can't use `Dummy_impl` class directly as it is not `exported` and this is done on purpose - to fully hide all the implementation details.

>    Dummy_impl inst;
>    inst.Dummy_Method(out);

To create an instance of you block you can use the static `make` method as define in `Dummy.h` and then call the virtual `Dummy_Method`, something like

```cpp
     gr::Dummy::sptr p = gr::Dummy::make();
     p->Dummy_Method(out);
```

nedsana, 2022-01-07:
Hi, 

Sorry for spamming the "issues" section. I was bothered by the "python and none-const reference" problem, described above, and I thought I found a problem. 

As for the C++ tests, I completely neglected the static make method. It worked fine and now I have a way to implement the tests I need.

Thank you.

closed 2022-01-07 by nedsana