gnuradio/gnuradio #5449: No pulse shaping of imaginary part in RRC with the constellation modulator

GillesC, 2022-01-07:
### What happened?

Strange behavior of the pulse shaping for the constellation modulator block.

### System Information

OS: Ubuntu 20.04
GR Installation Method: apt

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

9.9.4.0

### Steps to Reproduce the Problem

Create a QPSK/BPSK modulator block with the constellation modulator with any RRC parameters and check the imaginary and real part of the outputted complex signal.
I would expect that for the BSPK symbol 0 (non-differential), the imaginary part would be zero, and the real part would have a pulse (shaped according to the RRC) with a negative amplitude. However, the imaginary part is just -1, so no pulse shaping is applied.

willcode, 2022-01-07:
It's the total I/Q signal that matters, since they are not transmitted or processed separately.

closed 2022-01-13 by GillesC

aoweis, 2023-02-20:
Hi,
I have the same question and could not really understand the provided answer. For a zero symbol, the output of the constellation modulator should be:
Real: an RRC pulse with negative amplitude
Imaginary: zero.

However, the output as shown in GillesC question has a constant real part at -1. Why is not convolved with the RRC pulse shape?
I'm not sure I understand the given answer, the I and Q samples are combined in the RF signal but then separated again at the receiver baseband, and the decision of the value of the symbol is taken based on value of Real and Imaginary parts together.