gnuradio/gnuradio #5461: digital/map_bb ImportError specific to maint-3.9

ejohnson73, 2022-01-11:
### What happened?

Any import of digital will cause the following error. 

ImportError: $HOME/gnuradio39/lib/libgnuradio-digital.so.3.9.5: undefined symbol: _ZN2gr7digital11map_bb_impl10s_map_sizeE

It appears that **digital/map_bb: Add missing definition for a static class member #3677** needs to be merged into maint-3.9. 

### System Information

OS: Linux Ubuntu 18.04
GR Installation Method: Pybombs 
GCC: gcc version 8.4.0 (Ubuntu 8.4.0-1ubuntu1~18.04) 

### GNU Radio Version

3.9 (maint-3.9)

### Specific Version

3.9.5.0

### Steps to Reproduce the Problem

Run any example file in gr-digital. 
For example: 
`python3 ~/gnuradio39/share/gnuradio/examples/digital/demod/symbol_sync_test_float.py`

### Relevant log output

```shell
python3 ~/gnuradio39/share/gnuradio/examples/digital/demod/symbol_sync_test_float.py 
Traceback (most recent call last):
  File "$HOME/gnuradio39/lib/python3.6/site-packages/gnuradio/digital/__init__.py", line 19, in <module>
    from .digital_python import *
ImportError: $HOME/gnuradio39/lib/libgnuradio-digital.so.3.9.5: undefined symbol: _ZN2gr7digital11map_bb_impl10s_map_sizeE

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "$HOME/gnuradio39/share/gnuradio/examples/digital/demod/symbol_sync_test_float.py", line 32, in <module>
    from gnuradio import digital
  File "$HOME/gnuradio39/lib/python3.6/site-packages/gnuradio/digital/__init__.py", line 23, in <module>
    from .digital_python import *
ImportError: $HOME/gnuradio39/lib/libgnuradio-digital.so.3.9.5: undefined symbol: _ZN2gr7digital11map_bb_impl10s_map_sizeE
```

willcode, 2022-01-11:
The py file is generated from a grc file included in the source tree. If you regenerate the py file, does it work? It did for me.

`s_map_size` is now in the `.h` file, not the `.cc` file, so it is also possible that you have mixed versions installed.

`from gnuradio import digital` works here without a crash.

Please check for multiple installations or old shared libraries (including in python site/dist directories) and let us know how it goes.

ejohnson73, 2022-01-11:
During the compile I see the following. 

`[ 92%] Linking CXX shared library libgnuradio-trellis.so
[ 92%] Linking CXX shared module digital_python.cpython-36m-x86_64-linux-gnu.so
**libgnuradio-digital.so.3.9.5.0: warning: undefined reference to `gr::digital::map_bb_impl::s_map_size'**
[ 93%] Built target digital_qa_header_buffer.cc
**libgnuradio-digital.so.3.9.5.0: warning: undefined reference to `gr::digital::map_bb_impl::s_map_size'**
[ 93%] Built target digital_qa_header_format.cc
[ 94%] Built target gnuradio-trellis
[ 94%] Linking CXX shared module trellis_python.cpython-36m-x86_64-linux-gnu.so
[ 95%] Built target trellis_python
[100%] Built target digital_python`

`from gnuradio import digital` 
generates the error. 

Adding commit 3282fa7 fixes the issue for me. 

My exact version is 
`commit cb2ab8dd1555ec08a9afbfec675b6a7c8d9dbc90 (HEAD -> maint-3.9, tag: v3.9.5.0, origin/maint-3.9)`

Please let me know what else I can provide to help narrow down the differences between my system and the CI system.

willcode, 2022-01-11:
From the Ubuntu 18.04 build log for v3.9.5.0:
```
[ 99%] Building CXX object gr-digital/python/digital/bindings/CMakeFiles/digital_python.dir/python_bindings.cc.o
1950
[ 99%] Linking CXX shared module digital_python.cpython-36m-x86_64-linux-gnu.so
```
No related warnings, so I don't really have any ideas, other than something is cached on your system or another install is somehow interfering with the build.

ejohnson73, 2022-01-11:
[`GNU Radio 3.9 in Buildroot`](https://destevez.net/2021/10/gnu-radio-3-9-in-buildroot/) also shows the problem. When I get some time I'll try and make a Docker and see if it has the same issue. Then at least it will be repeatable. What about GCC versions? I have mine listed above. gcc version 8.4.0 (Ubuntu 8.4.0-1ubuntu1~18.04)

> After getting GNU Radio to build, I found a real oddity. The library libgnuradio-digital.so had an undefined symbol: gr::digital::map_bb_impl::s_map_size. This is a static constexpr that is only used in a few places in map_bb_impl.h and map_bb_impl.cc.

> Why g++ would create a symbol for this instead of just replacing the constant value in the generated code escapes me. I’ve tried to search for some clues and understand the subtleties of constexpr, but after finding some references to ODR-used and remembering this joke, I decided to replace the static constexpr by a #define macro.

willcode, 2022-01-12:
CI used 7.5.0. Here is the run. The CMake portion should have info on versions/config.

https://github.com/gnuradio/gnuradio/runs/4754087649?check_suite_focus=true

willcode, 2022-01-12:
Original issue for v3.8: #3670

ejohnson73, 2022-01-13:
I isolated the difference. I had the build type set to NoOptWithASM inside my PYBOMBS configuration.  I also tried compiling with `gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)` and the `NoOptWithASM` which caused a segfault when attempting a  `from gnuradio import digital` instead of the ImportError when compiled with `gcc version 8.4.0 (Ubuntu 8.4.0-1ubuntu1~18.04)`.

ejohnson73, 2022-01-13:
I needed to single step with no optimization. It was the best way I found inside the build system as the Debug build still has -O2 if I recall correctly. I had a similar issue with 3.7 with NoOptWithASM.

marcusmueller, 2024-01-31:
@mormj  I'd say this means `NoOptWithASM` is broken, probably not even by us but by compilers. I think **I** would call this an unsupported use case – and remove this build type from cmake/Modules/GrBuildTypes.cmake