gnuradio/gnuradio #5462: ControlPort Monitor Error

mwk088, 2022-01-11:
Hi,

When setting Control Port Monitor to True in a simple flowgraph, I get the error below and it exists out.

**Error:**
free(): invalid pointer

**Environment Information:**
GNURADIO
testing-support;python-support;post-install;doxygen;gnuradio-runtime;gr-ctrlport;* thrift;gnuradio-companion;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-channels;gr-qtgui;gr-uhd;gr-utils;gr_modtool;gr_blocktool
3.9.3.0-rc1

CentOS 8.4.0
Thrift 0.15.0

Any help would be greatly appreciated.

Thank you

Notou, 2022-05-11:
I observe the same issue with the GR 3.10.1.1 from the Ubuntu repository, that ships with thrift 0.13.

ArturMarekNowak, 2022-09-28:
I am observing the same problem with 3.10.1.1 on Ubuntu 22.04.1 LTS

marcusmueller, 2024-01-31:
I don't even have a ctrlport build at the moment, definitely warranting a dedicated look into the state of ctrlport + perf monitor

Notou, 2024-02-06:
I have observed ctrlport to work recently, in two ways, on debian based distributions (Debian and Mint):
- `apt install thrift-compiler python3-thrift libthrift-dev` before compiling from source (Tested with 3.10.8)
- `conda install thrift thrift-compiler` On a radio conda installation (Tested up to latest 3.10.9)

On top of that, the `gr-perf-monitorx` and `gr-ctrlport-monitor` require installing `networkx  pygraphviz matplotlib`, (from conda, pip, ...)