gnuradio/gnuradio #5463: `const_source` doesn't work with byte sources

ThomasHabets, 2022-01-11:
### What happened?

A bug happened!

When creating a byte const source, you get the error `Value 'gr.sizeof_char' not in allowed values` in the graph in grc.

### System Information

OS: Ubuntu 21.04
GR Installation Method: from source. Git commit 48e2780d0d84a06d1f22fd880d71595d779d77ce.

### GNU Radio Version

3.10-git (master)

### Specific Version

3.10.0.0-rc4

### Steps to Reproduce the Problem

1. Create const source
2. Set type to byte
3. Put some const in the const field. Like `1`
4. Observe error message

ThomasHabets, 2022-01-11:
Gah, it was my old enemy `~/.gnuradio/config.conf` overwriting where blocks are looked for. Again.

```
$ grep global .gnuradio/config.conf
global_blocks_path = /usr/share/gnuradio/grc/blocks:/usr/local/share/gnuradio/grc/blocks
```

Comment out that line and it works.

closed 2022-01-11 by ThomasHabets

mormj, 2022-01-12:
I hate that line!