gnuradio/gnuradio #5470: Windows Binaries not found

OfficialEsco, 2022-01-13:
Hey! WinGet Package Contributor here.

I just checked our package and the urls return 404 not found, the Installation guide link also returns 404 https://wiki.gnuradio.org/index.php/InstallingGR#Windows
Is this a temporary issue and i just had great timing, are they moved or are they not supported anymore?

Link to our issue: https://github.com/microsoft/winget-pkgs/issues/40715

marcusmueller, 2022-01-13:
Hi @OfficialEsco! Not quite sure, that link doesn't return a 404. To the best of my knowledge, there's not been working on the GNU Radio wiki webserver in the last 15 minutes. So, this might be a problem with your proxy?

OfficialEsco, 2022-01-13:
Weeeeird... i just got a confirmation on our end too.. Sorry for the confusion then >.<

closed 2022-01-13 by OfficialEsco