gnuradio/gnuradio #5476: FMComms 2/3/4 Sources not working

zhouzhiwen2000, 2022-01-15:
### What happened?

With FMComms 2/3/4 Sources, the grc file would return an error during generation.
Error message:
Generate Error: (NameError("'len_tag_key' is not defined"), 'iio.fmcomms2_sink_${type.type}(${uri}, [tx1_en, tx2_en], ${buffer_size}, ${cyclic})\nself.${id}.set_len_tag_key(${len_tag_key})\nself.${id}.set_bandwidth(${bandwidth})\nself.${id}.set_frequency(${frequency})\nself.${id}.set_samplerate(${samplerate})\nself.${id}.set_attenuation(0, ${attenuation1})\nself.${id}.set_filter_params(${filter_source}, ${filter}, ${fpass}, ${fstop})\n')

### System Information

OS: Ubuntu 20.04 LTS
GR Installation Method: Source

### GNU Radio Version

3.10-git (master)

### Specific Version

3.10.0.0

### Steps to Reproduce the Problem

Add an FMComms 2/3/4 source and click "Generate the flow graph" in the gnuradio companion.

zhouzhiwen2000, 2022-01-15:
update: After adding len_tag_key in the block.yml and fixing other minor issues, it still won't work.
It's reporting RuntimeError: Channel out of range for this device even though it has voltage1.
iio_device_find_channel(phy, "voltage1", false) seems to be returning wrong values.
iio_attr from the ZYNQ:
[image]

zhouzhiwen2000, 2022-01-15:
Also, grcs with the sink is reporting Segmentation fault.

zhouzhiwen2000, 2022-01-15:
In https://github.com/gnuradio/gnuradio/blob/master/gr-iio/lib/fmcomms2_source_impl.cc
line 336 and 352 
`bool is_fmcomms4 = !iio_device_find_channel(phy, "voltage1", false);`
This seems a little odd, since iio_device_find_channel returns NULL when the channel is not found.
http://analogdevicesinc.github.io/libiio/v0.18/group__Device.html#gacf9fe193b8183df86eab56df7da7361d

adamhorden, 2022-01-15:
Hello 👋 

Thank you for taking the time to report this bug 🐛 and the detailed steps to reproduce.
I am taking a look into this.

**Adam Horden**

zhouzhiwen2000, 2022-01-15:
> Hello 👋
> 
> Thank you for taking the time to report this bug 🐛 and the detailed steps to reproduce. I am taking a look into this.
> 
> **Adam Horden**

OK, these are what I've changed so far.
https://github.com/zhouzhiwen2000/gnuradio/commit/d62c3d7805f93328690d0e7dfb2cd96bf5b8a6dc

tfcollins, 2022-01-18:
I have a fix to address this. I need to finish up some testing but I'll put a PR together tomorrow.

-Travis

tfcollins, 2022-01-19:
@mormj Sources seem to be working fine with some minor fixes but there are issues with the volk calls on the sink blocks.

https://github.com/gnuradio/gnuradio/blob/1704a8f88b34daf224a5665dd55f8143832d876f/gr-iio/lib/fmcomms2_sink_impl.cc#L358 is causing segmentation faults with multiple loop iterations. Is there an alignment requirement on volk that is missing?

My branch updates so far: https://github.com/tfcollins/gnuradio/tree/tfcollins/fmcomms-fix

zhouzhiwen2000, 2022-01-19:
https://github.com/tfcollins/gnuradio/blob/046cd67052878c54303571dd671bcdbdf6e01615/gr-iio/grc/iio_fmcomms2_source.block.yml#L166
I am a little confused, won't this mean that rx1_en enables both rx1 and rx2 and rx2_en actually enables AuxADC?

tfcollins, 2022-01-19:
@zhouzhiwen2000 from the perspective of the IIO driver a channel is actually in reference to an individual ADC or DAC. So you need to enable pairs to get your complex signals. If you look at the IIO channels for cf-ad9361-lpc you will see voltage0->voltage3. voltage0/voltage1 is the first I/Q pair and voltage2/voltage3 is the second I/Q pair.

-Travis

zhouzhiwen2000, 2022-01-19:
> @zhouzhiwen2000 from the perspective of the IIO driver a channel is actually in reference to an individual ADC or DAC. So you need to enable pairs to get your complex signals. If you look at the IIO channels for cf-ad9361-lpc you will see voltage0->voltage3. voltage0/voltage1 is the first I/Q pair and voltage2/voltage3 is the second I/Q pair.
> 
> See how the channels are enabled/mapped: https://github.com/tfcollins/gnuradio/blob/master/gr-iio/lib/fmcomms2_source_impl.cc#L46
> 
> -Travis

That clears things up, thank you for explaining!

mormj, 2022-01-19:
> is causing segmentation faults with multiple loop iterations. Is there an alignment requirement on volk that is missing? My volk is a little rusty here but even if we set alignments won't that kinda break set_output_multiple?

I'm not sure if the missing alignment requirement would cause segfaults, or just make the volk calls less efficient.

This code looks problematic:
```
        if (noutput_items > (int)d_device_bufs[i].size()) {
            d_device_bufs[2 * i].resize(noutput_items);
            d_device_bufs[2 * i + 1].resize(noutput_items);
```
The if statement should probably be `d_device_bufs[2*i].size()

PR #5501 (2022-01-21): FMComms234 Source/Sink fixes
# Pull Request Details

PR fixes #5476 for both the FMComms234 source and sink blocks.

## Description

The fixes update the GRC files to use the correct APIs for 1RX/2RX 1TX/2TX configurations that interface with AD9361/AD9363/AD9364 based systems using the IIO drivers. The core iio_device_{source/sink} blocks have been updated to allow inherited blocks to better define tags to be generated/processed when using the packet_len APIs. Unrelated to #5476 was the incorrect behavior of the incorrect number of channels being processed for these tags. All of these have be [... cut]
closed 2022-01-21 by mormj