gnuradio/gnuradio #5487: QT GUI Eye Sink requires unnecessary value for trigger level

duggabe, 2022-01-15:
### What happened?

Upon execution of a flowgraph containing the https://wiki.gnuradio.org/index.php/QT_GUI_Eye_Sink a small pop-up window asks for the trigger level even though the trigger level is specified in the Parameters.

### System Information

OS: Ubuntu 20.04.3
GR Installation Method: Source

### GNU Radio Version

3.10-git (master)

### Specific Version

3.10.0.0-rc4

### Steps to Reproduce the Problem

A test for the QT_GUI_Eye_Sink is in https://pastebin.com/kU3ymmqa

drmpeg, 2022-01-16:
It just seems to be the way the Eye Sink is coded. To suppress the popup, use the ```Free``` trigger mode.

duggabe, 2022-01-16:
I would still classify this as a bug. What if every Instrumentation GUI did the same thing? It would be a major inconvenience and get fixed immediately!

drmpeg, 2022-01-16:
I agree it should be taken out. I wonder what the original designer (@ChristopheSeguinot) was thinking.

A little further down in the code, it does the same thing if it's in tag trigger mode and there's no tag specified. That makes sense.

Anyway, I think it's just a one line fix. Remove:

https://github.com/gnuradio/gnuradio/blob/master/gr-qtgui/lib/eyedisplayform.cc#L368

dl1ksv, 2022-01-23:
There should definitely be no request for a value  for the trigger level, if the controlpanel is deactivated for the eye sink. And removing the line like @drmpeg proposes fixes this issue.
Indeed , we could remove all the occurrences of d_tr_level_act.
But I wonder what a user expects , if he activates the controlpanel in the config tab.
At the moment a window pops up and you must enter only once  a value for the trigger value. You have a + and - button for level and delay, where you can increase or decrease these values while the flowgraph is running..
Why is the trigger level differently treated than the trigger delay.

When we remove

https://github.com/gnuradio/gnuradio/blob/3349d483a1f7e5db0eea042815d237810373857b/gr-qtgui/lib/eyedisplayform.cc#L368

no window pops up. Is this sufficient, or should there be an additional box to enter a numeric value directly for trigger level and trigger delay?
In my opinion it is adequate to set the initial values for level and delay in grc and change them by + and - in the running flowgraph.
But this should be discussed.

duggabe, 2022-01-24:
I think it should be the same as all the other Instrumentation GUIs. If the user wants to change the value while running, they can "middle mouse click" and make the change if they do not have the control panel active.

I agree with @dl1ksv to remove all the occurrences of d_tr_level_act.

PR #5516 (2022-02-10): gr-qtgui: Eye Sink remove unnecessary request for trigger value

---

# Pull Request Details

Fixes #5487
## Description

Removes unneeded PopupMenu d_tr_level_act and tr_delay_act
and corresponding functions.

## Which blocks/areas does this affect?

gr-qtgui eye sink

## Testing Done

Ran the example in #5487
## Checklist
commit 09c3c4fa (2022-02-10): gr-qtgui: Eye Sink remove unnecessary request for trigger value

Fixes #5487

Removes unneeded PopupMenu d_tr_level_act and tr_delay_act
and corresponding functions.
PR #5580 (2022-02-16): gr-qtgui: Eye Sink remove unnecessary request for trigger value (backport to maint-3.10)
Fixes #5487

Removes unneeded PopupMenu d_tr_level_act and tr_delay_act
and corresponding functions.

(cherry picked from commit 09c3c4fa4bfb1a02caac74cb5334dfe065391e3b)

Backport https://github.com/gnuradio/gnuradio/pull/5577
commit a8135eb8 (2022-02-16): gr-qtgui: Eye Sink remove unnecessary request for trigger value

Fixes #5487

Removes unneeded PopupMenu d_tr_level_act and tr_delay_act
and corresponding functions.

(cherry picked from commit 09c3c4fa4bfb1a02caac74cb5334dfe065391e3b)
closed 2022-02-10 by mormj