gnuradio/gnuradio #5510: cmake is missing checks for components required to build gnuradio

balister, 2022-01-22:
### What happened?

With a new F35 install, I build gnuradio by compiling and installing packages to clear cmake failures and turn on enough components to be useful. After a good cmake run, the compile failed. After installing the following packages the build worked and passed QA checks.

python3-zmq (for zmq block QA)
python3-devel (for Python.h)

### System Information

Fedora 35

### GNU Radio Version

3.11-git (main)

### Specific Version

HEAD

### Steps to Reproduce the Problem

See above

drmpeg, 2022-01-22:
The python3-zmq for QA issue has been around for as long as I can remember.

drmpeg, 2022-01-22:
On Ubuntu, python3-dev gets installed with libboost-all-dev (as part of libboost-python-dev).

willcode, 2022-01-25:
Help request - use chat.gnuradio.org or the mailing list for further help.

closed 2022-01-25 by willcode