gnuradio/gnuradio #5513: Ubuntu's apt won't install from GNU Radio PPAs

warnes, 2022-01-23:
### What happened?

### Issue

On Ubuntu 21.10, adding the GNU Radio PPTs generates a warning message:
```
$ sudo add-apt-repository ppa:gnuradio/gnuradio-releases
[sudo] password for warnes: 
Repository: 'deb http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu/ impish main'
Description:
PPA for GNU Radio Tagged Releases
More info: https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Adding deb entry to /etc/apt/sources.list.d/gnuradio-ubuntu-gnuradio-releases-impish.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/gnuradio-ubuntu-gnuradio-releases-impish.list
Adding key to /etc/apt/trusted.gpg.d/gnuradio-ubuntu-gnuradio-releases.gpg with fingerprint 3FC4DB82A1FA9AF51B5ADC3FB399F08D393C3C5C
Hit:1 http://us.archive.ubuntu.com/ubuntu impish InRelease
Get:2 http://us.archive.ubuntu.com/ubuntu impish-updates InRelease [110 kB]
Get:3 http://security.ubuntu.com/ubuntu impish-security InRelease [110 kB]    
Ign:4 http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu impish InRelease        
Get:5 http://us.archive.ubuntu.com/ubuntu impish-backports InRelease [101 kB]                                        
Err:6 http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu impish Release                                     
  404  Not Found [IP: 91.189.95.85 80]
Reading package lists... Done                            
E: The repository 'http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu impish Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Subsequently running `apt` also generates warnings:
```
$ sudo apt update
Hit:1 http://us.archive.ubuntu.com/ubuntu impish InRelease
Get:2 http://security.ubuntu.com/ubuntu impish-security InRelease [110 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu impish-updates InRelease [110 kB]                              
Ign:4 http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu impish InRelease                                   
Get:5 http://us.archive.ubuntu.com/ubuntu impish-backports InRelease [101 kB]
Err:6 http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu impish Release    
  404  Not Found [IP: 91.189.95.85 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu impish Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

#### Cause

It appears that the lack of a properly signed `Releases` file in the root of the PPA directories is the cause oof the issue.

#### Desired Resolution:

Add a properly signed `Releases` file in the root of the PPA directories.

#### Impact

Users of Ubuntu 21.10 cannot install from the official GNU Radio PPAs.   Users who are not paying close attention [NB: Like me!] will be very likely to install the obsolete versions of gnuradio from the main Ubuntu repository, leading to problems and increased support requests.

### System Information

OS: Ubuntu 21.10
GR Installation Method: GNU Radio PPA

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

See above.

warnes, 2022-01-23:
Workaround:  Use Ubuntu 20.04 instead.

warnes, 2022-01-23:
I attempted all of the documented solutions, including:
* Add the argument ` --allow-insecure-repositories` to `apt update`.
* Add the options `allow-insecure=yes` and/or `trusted=yes` to the source list file.
* Add `Acquire::AllowInsecureRepositories=true;` and `Acquire::AllowDowngradeToInsecureRepositories=true;` to /etc/apt/apt.conf.d/99insecure

None of these was successful.

mormj, 2022-01-24:
For some reason, the building process on launchpad servers for 21.04 or 21.10 is not finding the pybind11 installation, so nothing else builds.  I haven't put effort into figuring this out, since 22.04 is just around the corner.

mormj, 2022-01-24:
Thank you for reporting, but closing this as an issue since it doesn't relate explicitly to the GR codebase

closed 2022-01-24 by mormj