gnuradio/gnuradio #5521: grc: write out grc_version based on actual version

willcode, 2022-01-25:
### Feature Description

In grc files, `grc_version` is statically set to `3.8.0` in `grc/core/default_flow_graph.grc`.

Write out the actual version, in `a.b.c.d` form when saved by GRC.

### Feature Urgency

medium (would be nice to have in the near future)

marcusmueller, 2022-01-25:
I'd actually call this a bug!

dkozel, 2022-01-25:
Some quick notes about the areas that this would need to be implemented in.

An example of an in-tree example from maint-3.7 that has the GRC version in the XML.
https://github.com/gnuradio/gnuradio/blob/maint-3.7/gr-blocks/examples/peak_detector2.grc#L2

Where in the GRC XML->YAML converter it'd read the GRC version.
https://github.com/gnuradio/gnuradio/blob/942aff023d2c8152623aacb433ba2ce9a99ad3df/grc/converter/flow_graph.py#L25

Where the file_format is currently set:
https://github.com/gnuradio/gnuradio/blob/942aff023d2c8152623aacb433ba2ce9a99ad3df/grc/core/FlowGraph.py#L398

GR version is available in a flowgraph object in `self.parent.config.version` I'm pretty sure. The `FlowGraph` parent is `Platform` so in the actual save_flowgraph function it could be accessed as `self.config.version`.
https://github.com/gnuradio/gnuradio/blob/942aff023d2c8152623aacb433ba2ce9a99ad3df/grc/core/platform.py#L349

PR #5553 (2022-02-15): grc: store grc version info in metadata
---

# Pull Request Details

Saving a flowgraph does not save the grc version
as this info is not part of the schema.

This pr adds grc_version to the schema definition and
saves the grc version string when the fg is saved.

## Related Issue

Possibly
fixes #5521

## Which blocks/areas does this affect?

All grc files, works with files with and  without grc_version set.
## Testing Done
commit b0d3304e (2022-02-15): grc: store grc version info in metadata

Saving a flowgraph does not save the grc version
as it this info is not part of the schema.

This pr adds grc_version to the schema definition and
saves the grc version string when the fg is saved.

Possibly
fixes #5521
PR #5574 (2022-02-16): grc: store grc version info in metadata (backport to maint-3.9)
Saving a flowgraph does not save the grc version
as it this info is not part of the schema.

This pr adds grc_version to the schema definition and
saves the grc version string when the fg is saved.

Possibly
fixes #5521

(cherry picked from commit b0d3304e1b82821b60027858eb220bc513e2cf6f)

Backport https://github.com/gnuradio/gnuradio/pull/5553
commit 793c21ff (2022-02-16): grc: store grc version info in metadata

Saving a flowgraph does not save the grc version
as it this info is not part of the schema.

This pr adds grc_version to the schema definition and
saves the grc version string when the fg is saved.

Possibly
fixes #5521

(cherry picked from commit b0d3304e1b82821b60027858eb220bc513e2cf6f)
closed 2022-02-15 by mormj