gnuradio/gnuradio #5529: Add version info to .grc files

warnes, 2022-01-27:
### Feature Description

At the moment, there is no way for a user to determine which version of gnuradio-companion was used to create a flow graph.  This makes it difficult to diagnose problems with 'missing blocks' caused by using a flow graph with a different gnuradio-companion version/

Suggestion:  
* Add a gnuradio version field to .grc files
* Check the version when opening a .grc file
* Warn the user in the case of version mismatch
* Ideally, direct the user to a wiki page describing how to update a flow graph 

### Feature Urgency

medium (would be nice to have in the near future)

willcode, 2022-01-27:
Related to https://github.com/gnuradio/gnuradio/issues/5521.

Writing out and check a version should be easy enough, but determining whether anything needs to be done as a result is difficult.

warnes, 2022-01-27:
Useful things to do when version mismatch is discovered, starting with the simples:

1. Warn the user about a version difference
2. Provide a link to a wiki page listing potential issues
3. Inform the user about any blocks with name or API changes (e.g. pdu block relocation, additional options to WAV file sink).
4. Automatically migrate (some) blocks.

warnes, 2022-01-27:
Idea 3 could be implemented for in-tree blocks by maintaining a machine-readable 'changelog' table containing the block name, version number, and (description of) change.  For out of tree blocks, maintainers could also provide changelog table.  
Idea 4 would be substantially harder, and is probably not worth the effort.