gnuradio/gnuradio #5530: Bypassing Hier  blocks not possible

reald, 2022-01-27:
### What happened?

A bypassing Hier  block (Pad Source - Pad Sink) is not possible. 

[image]

"Source and destination block cannot be the same"

### System Information

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.10.0.0-5-g3349d483

### Steps to Reproduce the Problem

1. Create Hier Block in GRC with only an input and connect it to the output pad sink.
2. Compile and install
3. Use this hier block in flow graph.
4. Get "Source and destination block cannot be the same" error

In the generated python file of the Hier block you can find this:
```
        ##################################################
        # Connections
        ##################################################
        self.connect((self, 0), (self, 0))
```

### Relevant log output

```shell
Executing: /usr/bin/python3 -u /home/user/optsrctest.py

Traceback (most recent call last):
  File "/home/user/optsrctest.py", line 192, in <module>
    main()
  File "/home/user/optsrctest.py", line 170, in main
    tb = top_block_cls()
  File "/home/user/optsrctest.py", line 134, in __init__
    self.optionalsource_0 = optionalsource()
  File "/home/user/optionalsource.py", line 34, in __init__
    self.connect((self, 0), (self, 0))
  File "/usr/local/lib/python3.8/dist-packages/gnuradio/gr/hier_block2.py", line 39, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/lib/python3.8/dist-packages/gnuradio/gr/hier_block2.py", line 106, in connect
    self.primitive_connect(*args)
ValueError: connect: src and destination blocks cannot be the same
```

dkozel, 2022-01-27:
Bypassing inside of the hier block, I can see the issue and it does seem like there's a possible enhancement to look up the parent of each pad outside of the hier block to see if it's possible to bypass it.

I think that bypassing it at the next level up should be possible? In GRC that would be just bypassing the entire monolithic hier block. Does that work?

reald, 2022-04-25:
Yes, bypassing the hier block consisting out of one pad source and one sink only when used in a flow graph is possible.

But my point is a hier block containing only a pad source and a connected pad sink is technically correct. So there is no reason why this flow graph does not generate working code.

marcusmueller, 2024-01-31:
@reald I think the title of this issue might be a bit misleading, because GRC offers a function called "Bypass", which allows, well, to bypass a block. 

I'm changing the issue to title, could you please check it reflects what you meant well?

This is the issue:

> [image]
> 
> -------------
> 
> [image]
> 
> https://gist.github.com/marcusmueller/d65075242473dc2c402837029b23bc6e

Which I can reproduce; 

however, the error message here is exactly on point: The GNU Radio run time cannot connect (`void hier_block2_detail::connect(…)` ports on the same block; that would result in cyclic logic regarding who is the parent of the hierarchical block (`set_parent` in `hier_block2_detail`). 

There's no exception from that for the case when the two blocks are the same, current hier block (`d_owner`); we could probably add that.

But since that probably opens the door for a whole lot of logic bugs, I'd be a bit careful about that! If you need that, it's not impossible, but if it's not really needed, I think at this point in the GR3 lifecycle I'd be hesitant to add exceptions like that.

reald, 2024-02-01:
> I'm changing the issue to title, could you please check it reflects what you meant well?

Checked, ok.

marcusmueller, 2024-02-01:
@reald just to understand how to prioritize this: What's the use case for this kind of hier block?

reald, 2024-02-01:
I guess there is no real use case for an empty hier block. But beginners may build their first block without any functional content and fall into this trap.

marcusmueller, 2024-02-01:
In that case, let's revisit that when this goes wrong!

closed 2024-02-01 by marcusmueller