gnuradio/gnuradio #5532: Optional Pad Sources/Sinks in Hier blocks can only be used when not connected

reald, 2022-01-27:
### What happened?

In Hier blocks Pad sinks/sources can be marked as optional. But if they are connected to other blocks in the Hier block this signal processing chain is not optional anymore and requires input/output. So optional pad sinks/sources can only be used when nothing is connected, so they do not make sense at all. 
If an optional source is not connected, the whole chain inside the Hier block should be optional, too. Hier blocks are like libraries, they might support more features but they are not needed in every use case.

### System Information

OS: Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

GR Installation Method: Source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.10.0.0-5-g3349d483

### Steps to Reproduce the Problem

1. Create Hier block with different (independent) signal processing path
2. Mark one pad source as optional
[image]
3. Install Hier block
4. Use Hier block in flow graph, do not connect optional port (because it´s optional)
5. Get error

### Relevant log output

```shell
Traceback (most recent call last):
  File "/home/user/main_test.py", line 140, in <module>
    main()
  File "/home/user/main_test.py", line 120, in main
    tb.start()
  File "/usr/local/lib/python3.8/dist-packages/gnuradio/gr/top_block.py", line 100, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
RuntimeError: multiply_const_ff(2): insufficient connected output ports (1 needed, 0 connected)
```

dl1ksv, 2022-02-02:
Did you connect the output ? With the output connected to a null sink I get 
  File "/usr/local/gnuradio/lib64/python3.10/site-packages/gnuradio/gr/top_block.py", line 100, in start
    top_block_start_unlocked(self._impl, max_noutput_items)
RuntimeError: multiply_const_ff(8): insufficient connected input ports (1 needed, 0 connected)
which I think is correct as the Multiply block needs an input independently whether the pad source is optional or not.

reald, 2022-04-25:
Yes, same here. This means that it is not possible to insert a complete optional signal paths into a hier block. An optional pad source can only be used when nothing is connected inside the hier block. This does not make really sense.

I want to realize optional signal paths which are not used in every case. Imagine calculations for 2 or 3 receivers depending on how much hardware you have. If only 2 inputs are present optional pad 3 should be left open. Then the complete signal path for this should do nothing and save cpu time (and battery). Of course this is only possible if there are no cross dependencies between used and unused inputs.

marcusmueller, 2024-01-31:
> So optional pad sinks/sources can only be used when nothing is connected, so they do not make sense at all.

agreed.