gnuradio/gnuradio #5537: get this when trying to open gnuradio-companion , installed via port

oli123v, 2022-01-28:
### What happened?

this error appears 

### System Information

OS: macos 12.2
GR Installation Method: macports

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

just ran gnuradio-companion

### Relevant log output

```shell
ranslated Report (Full Report Below)
-------------------------------------

Process:               Python [52056]
Path:                  /opt/local/Library/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.9.10 (3.9.10)
Code Type:             ARM-64 (Native)
Parent Process:        zsh [51991]
Responsible:           Terminal [51989]
User ID:               501

Date/Time:             2022-01-28 22:33:05.6039 +0000
OS Version:            macOS 12.2 (21D49)
Report Version:        12
Anonymous UUID:        695F9D7B-6AE7-F666-95FC-57EF1E8B4A5A

Sleep/Wake UUID:       0697069B-2A5B-44A7-B0CD-99C7565E0EF9

Time Awake Since Boot: 6200 seconds
Time Since Wake:       5062 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x00000001ad34cb6c
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [52056]

Application Specific Information:
detected buffer overflow

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_c.dylib             	       0x1ad34cb6c __chk_fail_overflow + 24
1   libsystem_c.dylib             	       0x1ad2dbb30 __memmove_chk + 40
2   libgtk-3.0.dylib              	       0x119bec0ec _gtk_marshal_VOID__OBJECT_OBJECTv + 52
3   libgobject-2.0.0.dylib        	       0x103121458 _g_closure_invoke_va + 224
4   libgobject-2.0.0.dylib        	       0x1031366f0 g_signal_emit_valist + 852
5   libgobject-2.0.0.dylib        	       0x103136f3c g_signal_emit + 28
6   libgtk-3.0.dylib              	       0x1199c8460 gtk_css_node_reposition + 348
7   libgtk-3.0.dylib              	       0x119b0d9f8 gtk_style_context_save_named + 92
8   libgtk-3.0.dylib              	       0x119924590 gtk_style_update_from_context + 100
[... 420 lines cut ...]
  "experiments" : [

  ]
}
}

Model: MacBookPro17,1, BootROM 7429.81.3, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Display: Q3279WG5B, 1920 x 1080 (1080p FHD - Full High Definition), MirrorOff, Online
Memory Module: LPDDR4
AirPort: Wi-Fi, wl0: Oct 26 2021 16:10:41 version 18.20.310.15.7.8.120 FWID 01-5d6c3867
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: 2.4G Mouse
USB Device: USB Billboard Device
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

luzpaz, 2023-07-22:
Needs `macos` label

marcusmueller, 2024-01-31:
Lacking any apple hardware, I can't try to reproduce. Soo, adding the "help wanted" tag.