gnuradio/gnuradio #5554: Flowgraph shows "ERROR: READ LINE:-9 ERROR:READ INTEGER:-9" when is closed

smx12, 2022-02-04:
### What happened?

Every time I close the following flowgraph it shows these two messages:
`ERROR: READ LINE: -9
ERROR: READ INTEGER: -9`
```
options:
  parameters:
    author: User
    category: Custom
    cmake_opt: ''
    comment: ''
    copyright: '2022'
    description: FM Radio to UHF
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: qt_gui
    hier_block_src_path: '.:'
    id: fm_to_uhf
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: prompt
    sizing_mode: fixed
    thread_safe_setters: ''
    title: FM Radio PlutoSDR
    window_size: 1280, 1024
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [8, 4]
    rotation: 0
    state: enabled

blocks:
- name: fm_preset_station
  id: variable_qtgui_chooser
  parameters:
    comment: ''
    gui_hint: ''
    label: FM Station
    label0: La 100
    label1: Aspen
    label2: Mega
    label3: La Pop
    label4: ''
    labels: '[]'
    num_opts: '4'
    option1: int(102.3e6)
    option2: int(98.3e6)
    option3: int(107.3e6)
    option4: '4'
    options: '[0, 1, 2]'
    orient: Qt.QHBoxLayout
    type: int
    value: int(99.9e6)
    widget: radio_buttons
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [859, 24]
    rotation: 0
    state: true
- name: fm_station
  id: variable_qtgui_range
  parameters:
    comment: ''
    gui_hint: ''
    label: FM Station
    min_len: '200'
    orient: Qt.Horizontal
    rangeType: int
    start: int(88e6)
    step: int(50e3)
    stop: int(108e6)
    value: fm_preset_station
    widget: counter_slider
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [569, 21]
    rotation: 0
    state: true
- name: sample_rate
  id: variable
  parameters:
    comment: ''
    value: int(768e3)
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [189, 28]
    rotation: 0
    state: enabled
- name: uhf_freq
  id: variable_qtgui_range
  parameters:
    comment: ''
    gui_hint: ''
    label: UHF Freq
    min_len: '200'
    orient: Qt.Horizontal
    rangeType: float
    start: int(420e6)
    step: int(10e3)
    stop: int(470e6)
    value: int(466e6)
    widget: counter_slider
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [715, 22]
    rotation: 0
    state: true
- name: analog_nbfm_tx_0
  id: analog_nbfm_tx
  parameters:
    affinity: ''
    alias: ''
    audio_rate: int(32e3)
    comment: ''
    fh: '-1.0'
    max_dev: 5e3
    maxoutbuf: '0'
    minoutbuf: '0'
    quad_rate: int(sample_rate/decimation)
    tau: 75e-6
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [696, 480]
    rotation: 0
    state: true
- name: analog_wfm_rcv_0
  id: analog_wfm_rcv
  parameters:
    affinity: ''
    alias: ''
    audio_decimation: int(int(sample_rate/decimation)/32000)
    comment: ''
    maxoutbuf: '0'
    minoutbuf: '0'
    quad_rate: int(sample_rate/decimation)
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [469, 504]
    rotation: 0
    state: enabled
- name: audio_sink_0
  id: audio_sink
  parameters:
    affinity: ''
    alias: ''
    comment: ''
    device_name: default
    num_inputs: '1'
    ok_to_block: 'True'
    samp_rate: '32000'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [706, 394]
    rotation: 0
    state: disabled
- name: decimation
  id: parameter
  parameters:
    alias: ''
    comment: ''
    hide: none
    label: Decimation
    short_id: ''
    type: intx
    value: '2'
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [302, 24]
    rotation: 0
    state: enabled
- name: iio_pluto_sink_0
  id: iio_pluto_sink
  parameters:
    affinity: ''
    alias: ''
    attenuation1: '0'
    auto_filter: 'True'
    bandwidth: int(200e3)
    buffer_size: sample_rate
    comment: ''
    cyclic: 'False'
    filter: ''
    frequency: uhf_freq
    samplerate: sample_rate
    uri: uri
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [1194, 449]
    rotation: 0
    state: true
- name: iio_pluto_source_0
  id: iio_pluto_source
  parameters:
    affinity: ''
    alias: ''
    auto_filter: 'True'
    bandwidth: '200000'
    bbdc: 'True'
    buffer_size: sample_rate
    comment: ''
    filter: ''
    frequency: int(fm_station)
    gain1: '''manual'''
    manual_gain1: '64'
    maxoutbuf: '0'
    minoutbuf: '0'
    quadrature: 'True'
    rfdc: 'True'
    samplerate: sample_rate
    uri: uri
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [29, 424]
    rotation: 0
    state: true
- name: low_pass_filter_0
  id: low_pass_filter
  parameters:
    affinity: ''
    alias: ''
    beta: '6.76'
    comment: ''
    cutoff_freq: '48000'
    decim: decimation
    gain: '1'
    interp: '1'
    maxoutbuf: '0'
    minoutbuf: '0'
    samp_rate: int(sample_rate/decimation)
    type: fir_filter_ccf
    width: '44100'
    win: firdes.WIN_HAMMING
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [258, 464]
    rotation: 0
    state: enabled
- name: low_pass_filter_0_0
  id: low_pass_filter
  parameters:
    affinity: ''
    alias: ''
    beta: '6.76'
    comment: ''
    cutoff_freq: '5000'
    decim: '2'
    gain: '1'
    interp: decimation
    maxoutbuf: '0'
    minoutbuf: '0'
    samp_rate: int(sample_rate/decimation)
    type: interp_fir_filter_ccf
    width: '2000'
    win: firdes.WIN_HAMMING
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [972, 464]
    rotation: 0
    state: enabled
- name: qtgui_sink_x_0
  id: qtgui_sink_x
  parameters:
    affinity: ''
    alias: ''
    bw: int(200e3)
    comment: ''
    fc: int(99.9e6)
    fftsize: '1024'
    gui_hint: ''
    maxoutbuf: '0'
    minoutbuf: '0'
    name: VHF Signal
    plotconst: 'True'
    plotfreq: 'True'
    plottime: 'True'
    plotwaterfall: 'True'
    rate: '10'
    showports: 'False'
    showrf: 'True'
    type: complex
    wintype: firdes.WIN_BLACKMAN_hARRIS
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [993, 302]
    rotation: 0
    state: disabled
- name: qtgui_sink_x_0_0
  id: qtgui_sink_x
  parameters:
    affinity: ''
    alias: ''
    bw: int(200e3)
    comment: ''
    fc: fm_station
    fftsize: '1024'
    gui_hint: ''
    maxoutbuf: '0'
    minoutbuf: '0'
    name: FM Signal
    plotconst: 'True'
    plotfreq: 'True'
    plottime: 'True'
    plotwaterfall: 'True'
    rate: '10'
    showports: 'False'
    showrf: 'True'
    type: complex
    wintype: firdes.WIN_BLACKMAN_hARRIS
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [481, 311]
    rotation: 0
    state: disabled
- name: uri
  id: parameter
  parameters:
    alias: ''
    comment: ''
    hide: none
    label: URI
    short_id: ''
    type: str
    value: ip:192.168.2.1
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [431, 24]
    rotation: 0
    state: enabled

connections:
- [analog_nbfm_tx_0, '0', low_pass_filter_0_0, '0']
- [analog_nbfm_tx_0, '0', qtgui_sink_x_0, '0']
- [analog_wfm_rcv_0, '0', analog_nbfm_tx_0, '0']
- [analog_wfm_rcv_0, '0', audio_sink_0, '0']
- [iio_pluto_source_0, '0', low_pass_filter_0, '0']
- [low_pass_filter_0, '0', analog_wfm_rcv_0, '0']
- [low_pass_filter_0, '0', qtgui_sink_x_0_0, '0']
- [low_pass_filter_0_0, '0', iio_pluto_sink_0, '0']

metadata:
  file_format: 1
```

Any idea what they mean and why?

### System Information

OS: (Linux Ubuntu 20.04.3 LTS)
GR Installation Method: (native distro packages - apt)

### GNU Radio Version

3.8  (maint-3.8)

### Specific Version

3.8.1.0

### Steps to Reproduce the Problem

Close the flowgraph and the errors are displayed.

velichkov, 2022-02-04:
Hi @smx12,

Welcome to GNU Radio!

> - name: iio_pluto_source_0
>   id: iio_pluto_source

> ### Specific Version
> 3.8.1.0

How did you install `gr-iio` and which version? The `gr-iio` was added to `gnuradio` in version 3.10.0 and before that it was an OOT module.

> ERROR: READ LINE: -9 ERROR: READ INTEGER: -9

Are you able to reproduce this error if you replace `iio_pluto_source` and `iio_pluto_sink` with other source and sink blocks (noise source and null sink for example)?

velichkov, 2022-02-04:
Hi @smx12 

Both these errors are printed in the `libiio` library so it is probably better to ask there.

> ERROR: READ LINE: -9 

https://github.com/analogdevicesinc/libiio/blob/e1e9421feeb092a46e0b75934303c76a6e8ca827/iiod-client.c#L52

> ERROR: READ INTEGER: -9

https://github.com/analogdevicesinc/libiio/blob/e1e9421feeb092a46e0b75934303c76a6e8ca827/iiod-client.c#L724

smx12, 2022-02-04:
> Hi @smx12,
>
> Welcome to GNU Radio!
>
>
>
> How did you install `gr-iio` and which version? The `gr-iio` was added to `gnuradio` in version 3.10.0 and before that it was an OOT module.
>
>
> Are you able to [...]

Hi, I've installed gr-iio like so:

git clone -b upgrade-3.8 https://github.com/analogdevicesinc/gr-iio.git
cd gr-iio
cmake .
make 
sudo make install
cd ..
sudo ldconfig

If I replace the iio source and sink the error messages disapear. Thank!

marcusmueller, 2024-01-31:
gr-iio in this version was an OOT, so out of scope for this tracker

closed 2024-01-31 by marcusmueller