gnuradio/gnuradio #5556: Deprecating Clock Recovery MM

777arc, 2022-02-05:
https://github.com/gnuradio/gnuradio/blob/120268de56dbf1a1ab01e27c601cd7d1342afaf7/gr-digital/grc/digital_clock_recovery_mm_xx.block.yml#L3

If we are actually going to remove `Clock Recovery MM` we need to first have a 1:1 mapping of `Clock Recovery MM` params -> 'Symbol Sync' params, because plenty of folks are going to have no idea how to actually do the switch.

solomonbstoner, 2022-02-05:
We had an [inconclusive discussion ](https://github.com/gnuradio/gnuradio/pull/4969) about the mapping when I wanted to replace Clock Recovery MM with Symbol Sync on one of the hier block. The issue was - what to do with the `mu` parameter. Im not sure what has happened since

marcusmueller, 2024-01-31:
tagging as low prio, because this is only important for straightforward porting, and FGs using the deprecated block have gotten rare. Not a good "solution", but realistic…