gnuradio/gnuradio #5561: Notify user when GRC file was last opened in older GNU Radio

dkozel, 2022-02-10:
### Feature Description

Continuing from https://github.com/gnuradio/gnuradio/issues/5521

GRC files used to, and now will have in the future, the version of GRC they were saved with stored in the header. GRC should read this version number and notify the user when it last last opened in a significantly older version of GNU Radio that is expected to have possible changes in behavior (a change in `a` or `b` in the version number `a.b.c.d`).

### Feature Urgency

low (just an idea)

ei8fdb, 2022-02-17:
@dkozel I'd like to put some design work forward for this notification. Is this a likely scenario -

Bob the radio-astronomer has a grc file, black-hole.grc, that he's created on his desktop using GRC 4.0.

(In the headers GR version=4.0 will be stored in the header.)

Bob wants to test it on his old-trusty radio-astronomy rig which is running a GNU Radio 3.1.

The purpose of the proposed notification is to tell him that the GRC file was written in a much newer Gnuradio version. Because of this, the GRC behaviour might not be what he expects.

Is that correct?

I've got some questions to understand better:

- do we know what Bob's expectations are when he opens the file in GR 3.1?
- is there any advice for users if this happens? i.e. upgrade the grc version? use older blocks?
- are there any ideas what those possible behaviours might be, or be localised to? e.g. filters, math calculations?
- is there a way for them to verify if the behaviour is as expected. i.e. I'm guessing there's not a checksum
- are there particular logs/outputs they should check to verify it's working/or not as it should?

ei8fdb, 2022-02-21:
A good [usability heuristic when dealing with errors](https://www.nngroup.com/articles/ten-usability-heuristics/
) - helping users recognise, diagnose and recover from them is "Error messages should be expressed in plain language (no error codes), *precisely indicate the problem, and constructively suggest a solution.*"

When I worked on the pip project, we faced a similar issue providing solution advice where users ran into[ dependency resolution problems](https://pip.pypa.io/en/stable/topics/dependency-resolution/?highlight=conflict). Due to the complexity, we couldn't tell the user the exact cause of the conflict but we did provide them with[ some possible solutions](https://pip.pypa.io/en/stable/topics/dependency-resolution/?highlight=conflict#possible-ways-to-reduce-backtracking).

In order to indicate the problem and suggest a solution, I'm trying to list the most possible causes of these "possible changes in behavior" of grc files, and how to solve them (if known).

### Causes for possible change in grc file behaviour 
So far I've got, the grc file:

**contains core block(s) that have been removed from this version of GRC**
Possible fix: use replacement blocks (maybe block X, Y, Z)

**contains core block(s) that didn't exist on this version of GRC**
Possible fix: use replacement blocks (maybe block X, Y, Z)

**contains block(s) from OOT module(s) that are not installed**
Possible fix: install OOT module D 

**interfaces on block(s) have changed**
Possible fix: redesign flowgraph to account for different i/fs

Can anyone help with other possible causes? And sense check my possible fix suggestions. @dkozel @mormj. Others?

Gowri213, 2025-12-24:
If this issue is open, Could I take this up?

Gowri213, 2025-12-24:
@marcusmueller ok, Thankyou!

Gowri213, 2025-12-24:
@marcusmueller Thanks for the go-ahead!

I’ve traced the flowgraph load path to Platform.parse_flow_graph(). The .grc metadata already gets parsed there.

My plan is to:
1. read metadata.grc_version from the file
2. compare its major/minor version with self.config.version
3. emit a non-blocking warning via Messages.send() when they differ, indicating behavior may change

I’ll start with this minimal approach unless a UI dialog is preferred.