gnuradio/gnuradio #5590: Faiure of source build with pybombs recipe gnuradio-default with swig arguments

john-g8cqx, 2022-02-20:
### What happened?

build failed with swig argument errors flagged...
[transcript.txt](https://github.com/gnuradio/gnuradio/files/8103706/transcript.txt)

### System Information

ubuntu 20.04.3
GR Installation Method: pybomb
[transcript.txt](https://github.com/gnuradio/gnuradio/files/8103717/transcript.txt)
s

### GNU Radio Version

3.8  (maint-3.8)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

[transcript.txt](https://github.com/gnuradio/gnuradio/files/8103718/transcript.txt)

jmfriedt, 2022-02-22:
Same issue with Debian/testing (bookworm)

mormj, 2022-02-23:
Since this appears to be an issue with pybombs itself, please open the issue under github.com/gnuradio/pybombs

closed 2022-02-23 by mormj