gnuradio/gnuradio #5593: packaging dependency on 3.10 even after following specific instructions

AryehGielchinsky, 2022-02-20:
### What happened?

Installed on Ubunutu 20.04 with the following instructions

1. sudo add-apt-repository ppa:gnuradio/gnuradio-releases
2. sudo apt-get update
3. sudo apt install gnuradio
4. sudo apt install python3-pip
5. pip install packaging

got message that "packaging" was already installed (in anaconda).

I tried running the [first tutorial](https://wiki.gnuradio.org/index.php?title=Your_First_Flowgraph) but got an error about "packaging" not being available.

I'm able to import "packaging" from a python prompt.

I uninstalled gnuradio 3.10, and installed 3.9 instead, and the tutorial runs fine.

### System Information

OS: Ubunutu 20.04
GR Installation Method: native distro packages - apt

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

See "What happened?"

ryanvolz, 2022-02-23:
Hmmm, do you have `python3-packaging` installed via apt? (`sudo apt install python3-packaging`) I'd have thought that was a dependency. From your description, it sounds like you have a conda environment active by default, but GNU Radio doesn't use that and actually uses system-installed Python packages when installed from the PPA.

marcusmueller, 2022-03-20:
yes, please don't mix `pip` installs and system linux (matter of fact, `pip install` without a virtualenv, or `--user`, is probably not what you want to do!)

777arc, 2022-04-09:
Looks like this is an issue with the latest PPA, I updated the top of https://wiki.gnuradio.org/index.php?title=InstallingGR but seems a little messy to not include it

closed 2024-01-31 by marcusmueller