gnuradio/gnuradio #5602: gr-uhd not properly interacting with RFNoC?

horvathp, 2022-02-23:
### What happened?

I've got an X310 hooked up over 2x 10G SFP+ into an Intel NIC. Everything is fine when benchmarking using UHD:
$ ~/uhd/host/build/examples/benchmark_rate --channels 0,1 --rx_rate 200e6 --args="type=x300,addr=192.168.30.2,second_addr=192.168.40.2"
runs without issues. 

I create a flowgraph in gnuradio, consisting of 
"RFNoC RX Radio" -> "RFNoC DDC" -> "RFNoC RX Streamer" -> Null sink,
(DDC decimation = 1, i.e., 200 Msps output rate).
still everything looks fine, samples are disappearing in the digital nihil as expected.

If I'm stressing things by hogging the CPU (i.e., 
$ stress -c 12
), overruns start to appear, which is understandable. However,
- the overruns persist after shutting down the stress process,
- and more disturbingly, they also persist on any subsequent run of the same flowgraph, even in the absence of the stress process.

It looks like the stream does not get properly shut down using the STOP_CONTINUOUS command, and this hangs something in the FPGA?

A subsequent run of the benchmark_rate warns like this:
[WARNING] [RFNOC::GRAPH] One or more blocks timed out during flush!
However, if we let the benchmark_rate run to completion, it seems to shut down things nicely. After that, both the benchmark and the gnuradio flowgraph run without issues.

This issue has been verified on a completely different PC hardware, with a single 10G connection. The same "stress" does not permanently cripple the benchmark_rate, only the gnuradio flowgraph.

### System Information

OS: Ubuntu 20.04
GR Installation Method: source
UHD v4.1.0.5, 6bd0be9cda5db97081e4f3ee3127c45eed21239c
FPGA git hash: 26793b8

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.1.1-19-g9edfade8

### Steps to Reproduce the Problem

1. Check that your HW can keep up with 200M streaming:
`$ ~/uhd/host/build/examples/benchmark_rate --channels 0,1 --rx_rate 200e6 --args="type=x300,addr=192.168.30.2,second_addr=192.168.40.2"
`
2. run the flowgraph as presented above
3. start the stress process:
`$ stress -c 12`
4. observe overruns
5. stop the stress process and observe the overruns still there
6. stop the flowgraph, restart it and the overruns will be there
7. run the benchmark_rate tool again to witness the same problem, then try again to see it disappear.

### Relevant log output

```shell
$ ~/uhd/host/build/examples/benchmark_rate --channels 0,1 --rx_rate 200e6 --args="type=x300,addr=192.168.30.2,second_addr=192.168.40.2"

[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.1.0.5-0-g6bd0be9c
[00:00:00.000061] Creating the usrp device with: type=x300,addr=192.168.30.2,second_addr=192.168.40.2...
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[WARNING] [RFNOC::GRAPH] One or more blocks timed out during flush!
Using Device: Single USRP:
  Device: X-Series Device
  Mboard 0: X310
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: WBX-120 RX+GDB
  RX Channel: 1
    RX DSP: 1
    RX Dboard: B
    RX Subdev: Unknown (0xffff) - 0
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: WBX-120 TX+GDB
  TX Channel: 1
    TX DSP: 1
    TX Dboard: B
    TX Subdev: Unknown (0xffff) - 0

[00:00:03.665055733] Setting device timestamp to 0...
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[00:00:04.715901524] Testing receive rate 200.000000 Msps on 2 channels
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO[00:00:14.886292339] Benchmark complete.

Benchmark rate summary:
  Num received samples:     0
  Num dropped samples:      0
  Num overruns detected:    99
  Num transmitted samples:  0
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 0
  Num underruns detected:   0
  Num late commands:        0
  Num timeouts (Tx):        0
  Num timeouts (Rx):        0

Done!

hp@efop0:~/gr-xtremis/examples$ ~/uhd/host/build/examples/benchmark_rate --channels 0,1 --rx_rate 200e6 --args="type=x300,addr=192.168.30.2,second_addr=192.168.40.2"

[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.1.0.5-0-g6bd0be9c
[00:00:00.000115] Creating the usrp device with: type=x300,addr=192.168.30.2,second_addr=192.168.40.2...
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
Using Device: Single USRP:
  Device: X-Series Device
  Mboard 0: X310
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: WBX-120 RX+GDB
  RX Channel: 1
    RX DSP: 1
    RX Dboard: B
    RX Subdev: Unknown (0xffff) - 0
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: WBX-120 TX+GDB
  TX Channel: 1
    TX DSP: 1
    TX Dboard: B
    TX Subdev: Unknown (0xffff) - 0

[00:00:02.259833869] Setting device timestamp to 0...
[INFO] [MULTI_USRP]     1) catch time transition at pps edge
[INFO] [MULTI_USRP]     2) set times next pps (synchronously)
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
[WARNING] [0/Radio#1] Attempting to set tick rate to 0. Skipping.
[00:00:03.715127733] Testing receive rate 200.000000 Msps on 2 channels
[00:00:13.765913988] Benchmark complete.

Benchmark rate summary:
  Num received samples:     3999955600
  Num dropped samples:      0
  Num overruns detected:    0
  Num transmitted samples:  0
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 0
  Num underruns detected:   0
  Num late commands:        0
  Num timeouts (Tx):        0
  Num timeouts (Rx):        0

Done!
```

horvathp, 2022-03-17:
To give some more information, the error happens at sample rates as low as 50 Msps. I haven't tried lower than that yet. I think this is pretty problematic.

oguztudo, 2023-03-28:
Could you find a solution to this problem? I am having the issue that my usrp gives me a warning that it tries to set the tick rate to 0 which results in overruns.

cyjohnson-ss, 2023-05-31:
I'm seeing this same problem "One or more blocks timed out during flush!".  It seems to persist over stopping and restarting the stream.

cyjohnson-ss, 2023-05-31:
First thing that comes out when restarting the stream is a "D".

cyjohnson-ss, 2023-05-31:
Sorry for all the messages.  I just noticed when starting up, I get the "D" and also a Rx Seq Error.  Could this be due to the fact that the max stream packet size is 1996, yet I am requesting 2048.  In this case, maybe a "fragmented" packet is stuck in the FPGA?  .... just a though.

cyjohnson-ss, 2023-05-31:
More for you to chew on....
Once this warning occurred ("One or more blocks timed out during flush!), I couldn't make it go away no matter how many times I started it.
Started it with "sudo -k nice -n -19 su -c ./MyProgramName $USER" to ensure that the RX has priority on the CPU.
Then I removed the nice -19 and ran it at regular priority.  The warning went away, and doesn't come back (and the corresponding "D").
When I run the nice -n -19 again, the warning message returns.

cyjohnson-ss, 2023-05-31:
Solution to my problem.....
It turns out the "sudo -k nice -n -19 su -s ./MyProgramName $USER" with "-s" must be used instead of "-c", as when "Ctrl-C" was entered to terminate the streaming, it was actually caught by the shell and the shell terminated the process.  This means that the stream wasn't cleaned up properly so the following wasn't getting called:
    // Issue command to stop the stream
    stream_cmd.stream_mode = uhd::stream_cmd_t::STREAM_MODE_STOP_CONTINUOUS;
    rx_stream->issue_stream_cmd(stream_cmd);
The next time the program was called it generated the message "One or more blocks timed out during flush!" followed by an immediate "D".  Under this now reproducible circumstance, it seems like the UHD may still not behaving properly, but I will leave it up to developers to decide.
Here is output when error happens after restart (the bad case mentioned):

> sudo -k nice -n -19 su -c ./MyProgramName $USER
Creating the usrp device with: addr0=192.168.30.2 ...
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.4.0.cjohnson-fb-sdrx-68-g02558b69
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [X300] Radio 1x clock: 200 MHz
[WARNING] [RFNOC::GRAPH] One or more blocks timed out during flush!
Setting RX Rate: 200.000000 Msps... Actual: 200.000000 Msps... 
Setting RX Freq: 1223.000000 MHz... Actual: 1223.000003 MHz...
Setting RX Gain: 20.000000 dB...    Actual: 20.000000 dB...
Setting RX Bandwidth: 200.000000 MHz... Actual: 200.000000 MHz...
Checking RX... RXLO: locked ...
mboard 0: X310
Clock source is now gpsdo
Time source is now gpsdo
Waiting for ref_locked...USRP locked to 10MHz reference
Waiting for the GPSDO to warm up...The GPSDO is warmed up and communicating
GPS Locked

rx_stream->get_max_num_samps()=1996
Size of container=sizeof(buff)=24, # of buffs=2048, sizeof(std::complex<sample_t>)=4, total container + #buff*complex = 8216
Creating SHM GPU 8560 Ring Buffers each holding 2048 samples of size 4 bytes...
Total Size (Bytes) = 70123520
shm_GPU_buffer.capacity() = 8560
sizeof(ShmBuffers_t)=8192

Press Ctrl + C to stop streaming...
D

marcusmueller, 2024-01-31:
So, this seems to be solved :)

closed 2024-01-31 by marcusmueller