gnuradio/gnuradio #5603: Cannot install gnuradio, NumPy not found but is installed

GMVDavid, 2022-02-23:
I am attempting to install GnuRadio on Ubuntu 20.04.4 LTS. I am following the instructions at https://wiki.gnuradio.org/index.php/InstallingGR 
I have successfully installed Volk using the instructions provided, however when I try to run cmake with: 

cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../

I get the error: 

-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.6.5") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
CMake Error at cmake/Modules/GrPython.cmake:101 (target_link_libraries):
  Cannot specify link libraries for target "Python::NumPy" which is not built
  by this project.

However NumPy is installed and I can do import numpy in python 3.8. 

Within the last week I have installed GnuRadio using the same version of Ubuntu on a different workstation and the same instructions,  but it is failing on my laptop though all dependency versions appear to be the same for both.

Can anyone see what may be causing CMake to not see Numpy?  

[CMakeError.log](https://github.com/gnuradio/gnuradio/files/8123857/CMakeError.log)
[CMakeOutput.log](https://github.com/gnuradio/gnuradio/files/8123858/CMakeOutput.log)

mormj, 2022-02-23:
Try installing python3-packaging and re-running cmake.  This could be an unfortunate error message that has a different underlying cause.

GMVDavid, 2022-02-23:
I reinstalled python3-packaging and re-ran cmake but the error is the same. Do you know what the other underlying cause could be?

mormj, 2022-02-23:
I just opened up an ubuntu 20.04 docker and followed the https://wiki.gnuradio.org/index.php/InstallingGR#For_GNU_Radio_3.10,_3.9,_and_Main_Branch instructions from scratch, and could not reproduce that error

ryanvolz, 2022-02-23:
> Cannot specify link libraries for target "Python::NumPy" which is not built
> by this project.

This sounds like something that I've seen with an older version of CMake? But I thought we fixed it. Anyway, is there a chance you're using a different version of CMake?

mormj, 2022-03-01:
Closing this because it can not be replicated.

closed 2022-03-01 by mormj