gnuradio/gnuradio #5613: Companion is unresponsive after launch

Kllngii, 2022-02-27:
I am using a recent M1 Macbook with up to date firmware (macOS 12.2.1) and installed gnuradio via homebrew `brew install gnuradio`. After starting the companion using `gnuradio-companion&` I noticed some strange black blocks or sometimes a completly black window. However, it is still partly responsive, I am still able to double click blocks to open their properties and to open the save and load dialogs. The block selector on the righthand side and keyboard shortcuts are not working.

The strange thing is, that everything works fine until I save a file for the first time. If I later delete the saved .grc file everything is back to normal (except for small black boxes shown in the second picture)

[image]
[image]

kaiser-ren, 2022-03-04:
I have the same issue, I think the unusual behavior was started while I tried to connect the different components. 
My machine is MacBook Pro M1, python is 3.9.10, gnuradio is 3.9.3.0. 

I also noticed that below warning was appeared in the terminal, I below my machine has matplotlib. 

_Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.9.3.0 >>>

Block paths:
	/opt/homebrew/share/gnuradio/grc/blocks
gr::log 2022-03-04 22:14:11,081 :WARN: Matplotlib is a required dependency to use DistanceRadar and AzElPlot.  Please install matplotlib to use these blocks (https://matplotlib.org/)_

Kllngii, 2022-03-04:
I also got that slow YAML loading warning even though I‘ve installed libyaml with `brew install libyaml`.

kaiser-ren, 2022-03-05:
Provide more clues on this issue.  
I have another MacBook, it's not Apple Silicon, old one. I had installed gnuradion before and updated it to 3.9.3.0 this week. This MacBook had the same issue I described in the previous reply. 
However, after:
brew uninstall gnuradio
brew cleanup gruradio
brew install gnuradio

Gnuradio works fine on this old fashion MacBook. I will try this on my Apple Silicon MacBook. This is the screenshot when everything works well. 
[image]

kaiser-ren, 2022-03-05:
> Provide more clues on this issue. I have another MacBook, it's not Apple Silicon, old one. I had installed gnuradion before and updated it to 3.9.3.0 this week. This MacBook had the same issue I [...]

I just tried above three commands on MacBook (Apple Silicon), this method doesn't work and the issue is still the same like @Kllngii mentioned before.

kaiser-ren, 2022-03-07:
I also tried sudo port install gnuradio, the installation was successful, however, while I tried to launch gnuradio, the application didn't even pop-up.

wlopez-enkoding, 2022-03-23:
It’s weird because I don’t have a M1 chip but I have the same problem 🤔

kaiser-ren, 2022-03-23:
> It’s weird because I don’t have a M1 chip but I have the same problem 🤔

Sound like a common issue, I hope certain expert can give a workaround for this issue, or a timeline to fix. 
Appreciated in advance.

wlopez-enkoding, 2022-04-01:
Hey guys I came to tell you that in my case it was a problem that in the last versión in master branch was fixed. In short (for now) I had to turn scaling off in my Mac and that fixed the problem, @Kllngii maybe you should check out this option 🤷🏻‍♂️ C Ya !

justacec, 2022-04-01:
@wlopez-enkoding Do you mean the "main" branch or the "master" branch (for clarification).  Also, how did you disable the scaling?

wlopez-enkoding, 2022-04-03:
> @wlopez-enkoding Do you mean the "main" branch or the "master" branch (for clarification). Also, how did you disable the scaling?
> 
> edit: found that possibly setting "QT_AUTO_SCREEN_SCALE_FACTOR"=1 in the env might do the trick

yeah main branch. You can disable scaling in system preferences.

[image]

justacec, 2022-04-03:
But that would disable scaling for everything right.  The env bar seems to do it just for the app.

wlopez-enkoding, 2022-04-03:
> But that would disable scaling for everything right. The env bar seems to do it just for the app.
> […](#)

env didn't work for me 😕

ryanvolz, 2022-04-05:
> Hey guys I came to tell you that in my case it was a problem that in the last versión in master branch was fixed. In short (for now) I had to turn scaling off in my Mac and that fixed the problem, @Kllngii maybe you should check out this option 🤷🏻‍♂️ C Ya !

Let me see if I have this right. What I'm reading sounds like another manifestation of the bug that was fixed with https://github.com/gnuradio/gnuradio/pull/5111. Are all of these statements correct?:

1. Everyone experiencing this issue is using GNU Radio 3.9.3 or earlier on a machine with a "Retina" display (high DPI).
2. The problem goes away if you disable resolution scaling for the display.
3. The problem also goes away if you use GNU Radio built from the main branch (or really 3.9.4 or later, because it includes the fix linked above).

For everyone still having this problem and are stuck with GR 3.9.3, try making this change in the installed python code: https://github.com/gnuradio/gnuradio/pull/4814#issuecomment-925190019.

justacec, 2022-04-05:
Hate to break it.   I had the issue after building from the main branch the other day.  I set the env var to disable the scaling, seems like it worked.

ryanvolz, 2022-04-06:
> Hate to break it. I had the issue after building from the main branch the other day. I set the env var to disable the scaling, seems like it worked.

Interesting. You set `QT_AUTO_SCREEN_SCALE_FACTOR=1` in the environment, and not the global scaling factor, and that fixed it? GRC is gtk-based, so the QT environment variable shouldn't have any effect as far as I can tell, but the global setting would.

wlopez-enkoding, 2022-04-07:
> > Hey guys I came to tell you that in my case it was a problem that in the last versión in master branch was fixed. In short (for now) I had to turn scaling off in my Mac and that fixed the problem, @Kllngii maybe you should check out this option 🤷🏻‍♂️ C Ya !
> 
> Let me see if I have this right. What I'm reading sounds like another manifestation of the bug that was fixed with #5111. Are all of these statements correct?:
> 
> 1. Everyone experiencing this issue is using GNU Radio 3.9.3 or earlier on a machine with a "Retina" display (high DPI).
> 2. The problem goes away if you disable resolution scaling for the display.
> 3. The problem also goes away if you use GNU Radio built from the main branch (or really 3.9.4 or later, because it includes the fix linked above).
> 
> For everyone still having this problem and are stuck with GR 3.9.3, try making this change in the installed python code: [#4814 (comment)](https://github.com/gnuradio/gnuradio/pull/4814#issuecomment-925190019).

You're right, this is my workaround due to my brew installation.

wlopez-enkoding, 2022-04-07:
> Interesting. You set `QT_AUTO_SCREEN_SCALE_FACTOR=1` in the environment, and not the global scaling factor, and that fixed it? GRC is gtk-based, so the QT environment variable shouldn't have any [...]

What didn't work for me was setting a OS Global Environment Variable, I was wrong ?

ryanvolz, 2022-11-10:
I'm unclear if people here are still having problems, but: any chance https://github.com/gnuradio/gnuradio/pull/6352 fixes this? That's a python-only fix, so you can manually edit the appropriate installed python files to test it.

marcusmueller, 2024-01-31:
since we haven't heard from this in a year after asking whether the problem persists: closing

closed 2024-01-31 by marcusmueller