gnuradio/gnuradio #5625: Support installation via macPorts on macOS

nospam2000, 2022-03-05:
### What happened?

I wanted to use gnuradio via macports. This has been working in an older version. With my updated environment (macOS 11.6.3, Xcode 13.2.1, python 3.9 or 3.10) I haven't found a way to get it running.

Why not using homebrew? Mixing homebrew and macport is probably not a good idea and I'm already using lots of macports tools.

Here this list of things I tried:

1. macports installation of gnuradio @3.8.5.0_1 works and I can run gnuradio-companion, but when executing the graph I get the following error: `Class QMacAutoReleasePoolTracker is implemented in both /opt/local/libexec/qt5/lib/QtCore.framework/Versions/5/QtCore (0x10740c350) and /opt/local/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/PyQt5/Qt5/lib/QtCore.framework/Versions/5/QtCore`

No matter what I did, the installation always breaks my system by installing two conflicting versions of Qt5. This even broke other python apps like urh. 

I googled for hours but didn't find a solution.

2. macports installation of gnuradio-next @20200419-03d92ac1_7 fails because it doesn't compile. I fixed 3 compile problems and then gave up because investing so much time to an old version is not worth it.

3. installing gnuradio 3.10 from source
The instructions are outdated. They reference Python 2.7 and there is no macport port named 'gnuradio-devel' any longer.
The latest update in the repo with the build script is two years old and references gnuradio 3.8.

4. PyBOMBS
I tried that also some weeks ago, but also failed

You can see the story of using gnuradio on macOS  is a story of many failures. It would be really nice if you could support the people from macports to provide a version which can be used on macOS.

### System Information

OS: Mac: macOS macOS 11.6.3, Xcode 13.2.1, macports python 3.9 or 3.10
GR Installation Method: macports and PyBOMBS

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

```
sudo port install gnuradio-next
```

or

```
sudo port install gnuradio
```

### Relevant log output

```shell
Please have a look at all the issues for gnuradio at the macports bug tracking system:
https://trac.macports.org/query?status=!closed&port=~gnuradio
```

zeetabit, 2022-05-06:
The same problem. 
The brew version also is outdated.

Have issues with GUI on MacBook 16 that are probably solved on newer versions but cannot compline on latest OSX due to:
```
[DEBUG] Package python-click-plugins is not installable.
[ERROR] Package has no install method: python-click-plugins
[DEBUG] Unresolved install path.
```
The fix https://github.com/gnuradio/gr-recipes/pull/178/files not helped.

So can't try the latest version anyway.

marcusmueller, 2024-01-31:
excellent support via anaconda now, and macports was our go-to mac deployment so long! 

https://wiki.gnuradio.org/index.php?title=InstallingGR#Quick_Start

closed 2024-01-31 by marcusmueller