gnuradio/gnuradio #5627: gr-qtgui /gr-fec gnuradio/gr-fec/examples/ber_curve_gen.grc fails to build

dl1ksv, 2022-03-06:
### What happened?

Trying to build the gnuradio/gr-fec/examples/ber_curve_gen.grc fails as the qtgui_ber_sink fails to evaluate.

The error message is:

Param - Bus_Structure_Sink(bus_structure_sink):
	Value "list(map(lambda b:list(map(lambda a:b * len(esno) * 2 + a, range(len(esno)*2))), range(num_curves)))" cannot be evaluated:
	name 'num_curves' is not defined

but num_curves is set to 1 in the example.

### System Information

OS: Fedora 35 , build from scratch

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

Just open [ber_curve_gen flowgraph ](https://github.com/gnuradio/gnuradio/blob/main/gr-fec/examples/ber_curve_gen.grc) in grc

drmpeg, 2022-03-06:
Just FYI, there is a flow graph test program here.

https://github.com/mormj/gnuradio/blob/test_examples/grc/tests/test_examples.py

Run it with ```python3 -B -m pytest grc/tests -m "examples"``` in your source directory.

You can see that quite a few examples have been blacklisted (including the BER stuff).

It hasn't been merged yet. https://github.com/gnuradio/gnuradio/pull/5188

dl1ksv, 2022-03-07:
Thanks for this hint. I came accros this problem, when I was looking for a test case for the ber_curve_sink gui.
Meanwhile I found out that referencing a parameter of the block while setting another parameter of the block does not work at the moment.
That is: even if num_curves is set to 1 you can't use num_curves to set the value for bus_structure_sink.

But if you add a variable block named num_curves and set it to 1 and add another variable block named esno and set it's value to esno_0, the flowgraph can be build and run

marcusmueller, 2024-01-31:
Duplicate of #5198

closed 2024-01-31 by marcusmueller