gnuradio/gnuradio #5632: wrong font color on textboxes whihle using gtk dark theme

rodan, 2022-03-09:
### What happened?

my ~/.config/gtk-3.0/settings.ini contains

[Settings]
gtk-icon-theme-name = Adwaita
gtk-theme-name = Adwaita
gtk-font-name = DejaVu Sans 11
gtk-application-prefer-dark-theme = true

which means that all the fonts inside the Properties dialogs end up white. including the writing in the textedit boxes that have a custom background.  and due to the fact that this background is already very lightly colored I end up with while writing on white background in the case of #dtype_raw for instance. it's also almost impossible to read the white on light-green and light-gray textboxes.

so there are two bugs here:

1. even if I got 'gtk-application-prefer-dark-theme = true' and I selected 'qss = /usr/share/gnuradio/themes/dark.qss',
 GRC always ends up loading the 'LIGHT_THEME_STYLES' - I tested this by modifying grc/gui/canvas/colors.py.

2. the font color for all the #dtype_* fields should be just as customizable as the background of those fields. since the background is heavily modified by the application there is no way the gtk default dark theme will provide a font color fit for the light-colored custom textboxes this application uses.

[image]

### System Information

OS: Gentoo Hardened amd64 - stable
GR install - 
gnuradio-3.8.5.0 from the Gentoo portage tree

### GNU Radio Version

3.8  (maint-3.8)

### Specific Version

3.8.5.0

### Steps to Reproduce the Problem

start gnuradio-companion, go to Variable -> Properties and not see any values

nils-werner, 2022-04-02:
This is likely because of the code using colors like `red` instead of [named colors](https://gnome.pages.gitlab.gnome.org/libadwaita/doc/1-latest/named-colors.html) like `@error_bg_color`.

kreegahbundolo, 2023-11-12:
I see there are mentions of this issue being fixed in #2374 and #6389 but I just installed gnuradio from the ubuntu system apt repos on a clean Ubuntu 23.10 and when using dark mode the text is white and pretty unreadable. Maybe revisit the fixes to see what's up? The ubuntu system repos are giving me grc 3.10.7.0 so it should be recent enough

marcusmueller, 2023-11-12:
This is solved by now, but not in 3.10.7.0

closed 2023-11-12 by marcusmueller

s21jb, 2025-02-07:
Problem persists in 3.10.7.0 white text on white background in text boxes. Using dragonOS.