gnuradio/gnuradio #5635: Fail to build with LTO enabled

r-value, 2022-03-10:
### What happened?

Making package for Arch Linux, with LTO enabled. `ld` complains that:

```
/usr/bin/ld: libgnuradio-digital.so.3.9.4.0: undefined reference to `gr::digital::map_bb_impl::s_map_size'
```

### System Information

OS: Arch Linux
GR Installation Method: build from source
GCC version: 11.2.0

### GNU Radio Version

3.9  (maint-3.9)

### Specific Version

3.9.4.0

### Steps to Reproduce the Problem

Build script:

```
  export PYTHON=python3
  cd "$pkgbase-$pkgver"
  cmake -DCMAKE_INSTALL_PREFIX=/usr \
    -DPYTHON_EXECUTABLE=$(which python3) \
    -DPYTHON_INCLUDE_DIR=/usr/include/python3.10 \
    -DPYTHON_LIBRARY=/usr/lib/libpython3.10.so \
    -DGR_PYTHON_DIR=/usr/lib/python3.10/site-packages \
    -DENABLE_INTERNAL_VOLK=OFF \
    -DENABLE_GRC=ON \
    -DENABLE_GR_QTGUI=ON \
    -DQWT_LIBRARIES=/usr/lib/libqwt.so \
    -Wno-dev \
    -B build \
    -S .
  make -C build
```

### Relevant log output

```shell
https://gist.github.com/r-value/5b6fc6e54cdd92b34fce4128ebb3b8ff
```

rear1019, 2022-03-11:
https://github.com/gnuradio/gnuradio/blob/7c989ae932357801d1eadbf68eef1fdb3b3fbfd8/gr-digital/lib/map_bb_impl.h#L23

This declaration alone (without further code) only works since C++17; GNU Radio 3.9.4.0 is compiled using C++14. I wonder why it works at all, but this might be just another subtlety in C++… Anyway, the following patch should fix the issue for older C++ revisions (not tested).

```diff
diff --git a/gr-digital/lib/map_bb_impl.cc b/gr-digital/lib/map_bb_impl.cc
index d1f5b134f..5270bd891 100644
--- a/gr-digital/lib/map_bb_impl.cc
+++ b/gr-digital/lib/map_bb_impl.cc
@@ -18,6 +18,8 @@
 namespace gr {
 namespace digital {
 
+constexpr size_t map_bb_impl::s_map_size;
+
 map_bb::sptr map_bb::make(const std::vector<int>& map)
 {
     return gnuradio::make_block_sptr<map_bb_impl>(map);
```

marcusmueller, 2022-03-20:
Calling @ThomasHabets on this one!

ThomasHabets, 2022-03-20:
Yeah, @rear1019 is right. This is required in C++14 "if it is odr-used" (12.2.3.2 (class.static.data) paragraph 3), and actually deprecated as unnecessary in C++17 (D1 (depr.static_constexpr)).

I don't quite understand why LTO makes it ODR-used. My reading of 6.2 (basic.def.odr) paragraph 3 is that this should fall under the exception as it seems to me that the lvalue-to-rvalue conversion should yield a constant expression. I mean it *is* a constant expression, and we're using it just as-is.

In any case adding the definition, as @rear1019 said, can't hurt, and should help. Because from the logs it sure looks like the linker expects it.

r-value, 2022-03-26:
Seems that the patch from @rear1019 works. I'd appreciate it if this patch could be merged and released.

I'd like to leave this issue open until the changes are released.

marcusmueller, 2024-01-31:
solved by std upgrade in 3.10

closed 2024-01-31 by marcusmueller