gnuradio/gnuradio #5636: Zoom in and out centered at mouse position

Semnodime, 2022-03-11:
### Feature Description

- When zooming via <kbd>Ctrl</kbd>+<kbd>Mouse Wheel</kbd>:
Instead of zooming by stretching the plane from the (0,0) point, I'd kindly suggest to have zooming in and out centered with respect to the current mouse coordinate, as it would make most zooming related scrolling activity obsolete.

- When zooming via <kbd>Ctrl</kbd>+<kbd>+</kbd>:
This can stay as it is, as I am not sure changing it would positively impact the workflow of most users. I am open for conviction.

### Feature Urgency

medium (would be nice to have in the near future)

cgoxo, 2022-04-18:
Can I work on this issue? Need to know more about this.

BryantArias, 2022-09-20:
I would like to take on this feature as my first contribution. Would you be able to point me to where this would be implemented?

tofilwiktor, 2022-10-22:
I can try working on this

dev-DTECH, 2023-03-30:
@tofilwiktor are you still working on this? 
I would like to work on this issue.

dkozel, 2023-04-07:
@dev-DTECH Please go ahead!

luzpaz, 2023-08-10:
Any progress on this ?

Ansha4, 2024-01-07:
I want to work on this issue. @Semnodime can you please provide more info regarding it?

vnydv21, 2024-02-21:
Hi there, 

While using the flowgraph designer GUI, I realized that this feature is quite handy. Moreover, we can add a negative position direction scrolling (say one wants to put blocks before the (0,0) drawing axis), especially for crowded designs.
Overall, **it allows the designer to place blocks anywhere and zoom locally.**

Given, no implementation of this yet, I tried analysing the codebase, and I think we need to update the following stuff to get this done:

1. In these lines of code @ [gnuradio/grc/gui/DrawingArea.py](https://github.com/gnuradio/gnuradio/blob/039f0b89abfd2c37d493e59239c05b6c45eec21f/grc/gui/DrawingArea.py#L183) - whenever, the `mouse-position` is negative (< 0) nothing gets updated. Probably because the `adj.set_value()` has a lower limit of `0.0`.

   So, could try changing the lower limit, or can set the adjuments such that the page size increases and child widgets pan based on negative mouse position.

2. **Regarding the mouse centered zoom**: a naive approach can be to increment the page-size (utilizing the previous _negative position scrolling_) and pan widgets by an offset determined by the `widget-position` or `mouse-position` focused at.

Given, I am new to Open Source Contribution, please do suggest a better approach if possible.
Also, would like to work on the implementation suggested.

Cheers