gnuradio/gnuradio #5648: Problem about including ".h" in OOT block.

Zhang980823, 2022-03-18:
My OOT uses some self-defined .h files, or other GR modules' .h files , and then I can run make and install the module to gnuradio successfully, but I can't run the flow diagram. It occus error below:

ImportError: /usr/local/lib/libgnuraduo-mytest.so.0.0.1: undefined symbol: fec_encode()

It seems .o (dynamic library) has a problem. According to the experience of others on the Internet, it need to modify the cmakelist file to add the .h include path. I tried but no effect.
(Ubuntu 21.04 and Gnuradio v3.10.0.0git)

Thank you for your apply very much!

velichkov, 2022-03-18:
Hi @Zhang980823,

> My OOT uses some self-defined .h files, or other GR modules' .h files , and then I can run make and install the module to gnuradio successfully, but I can't run the flow diagram. It occus error below:
> 
> ImportError: /usr/local/lib/libgnuraduo-mytest.so.0.0.1: undefined symbol: fec_encode()

It is not enough to only include the header file (.h) unless it is a header only library (GR modules are not header only) or the function is defined/implemented in the header. You need to `link` the library where the function is implemented. 

In your toplevel `CMakeFiles.txt` you need to add
```cmake
find_library(MY_FEC_LIBRARY my_fec_library)
```
and in `lib/CmakeFiles.txt` in the `target_link_libraries` list add `MY_FEC_LIBRARY` variable
```cmake
target_link_libraries(gnuradio--mytest  
...
MY_FEC_LIBRARY
...
)
```

P.S.
Please look for help on chat.gnuradio.org or the mailing list. The issue tracker is used for bug reports and feature requests.

willcode, 2022-03-18:
Closing, since this is a help request.

closed 2022-03-18 by willcode