gnuradio/gnuradio #5650: Selector block does not consume inputs sufficiently

matt-g-87, 2022-03-19:
### What happened?

Hello all,
I have been using the (new) Selector block from gr-blocks for a while. In my case I have multiple inputs with a single output. _My inputs do not necessarily produce items at the same rate._ Sometimes I get hangs and the selector block stops performing the general_work function. After adding some debug logs, I have determined that this is caused by the general_work function not consuming all inputs sufficiently. What I mean by this, is that the non-selected inputs can build up large numbers of unconsumed items, which eventually leads to a hang. 

This issue appears to be caused by the use of `consume_each(noutput_items)` at the end of the general_work function in the selector block. This consume_each call, of course, tries to consume an equal number of samples every time from all inputs. This is based on the presumption that all of the inputs to the selector block come in at the same rate. However, since the changes made by the commit for issue #3762 and the solution #3763, the selector should no longer assume that inputs have the same rate of items being supplied to it (if any at all). The primary change of #3763 was that forecasting is only done for the selected input (not all inputs, as before that change).

Likewise, instead of calling consume_each(noutput_items), we should consider:
* Consuming noutput_items ONLY from the selected input
* And then consuming ALL ninput_items from each of the non-selected inputs. This clears out all the items in for the other, non-selected inputs. This prevents them from "building up" if they are provided to the selector faster than the selected input.

Since the intent of the Selector block is to pass along the selected input, and to drop all other inputs... then lets drop the other inputs as fast as possible.

I have tested replacing the `consume_each(noutput_items` call with something like this with good success:
```
for (unsigned i = 0; i < d_num_inputs; i++) {
    if (i == d_input_index)
        consume(i,noutput_items);
    else
        consume(i,ninput_items[i]);
}
```

This will consume all of the irrelevant/non-selected inputs as fast as possible.

### System Information

OS: Ubuntu 20.04 LTS
GR Installation Method: Source

### GNU Radio Version

3.9  (maint-3.9)

### Specific Version

3.9.5.0

### Steps to Reproduce the Problem

* Create a selector block with multiple inputs and one output
* On the "selected" input, provide items from a block at some rate _R_
* On the "non-selected" inputs, provide items from a different block at some rate much faster than _R_
* Observe an eventual hang, since the non-selected inputs aren't consumed fast enough

willcode, 2022-03-19:
This could be an option in the Selector. We'd have to think about whether it could cause deadlocks in other parts of a flowgraph. Most (all?) multiple-input blocks assume that all inputs are at the same rate.

jacobagilbert, 2022-03-19:
This is definitely expected behavior, the selector is only useful in specific cases. I wonder if this functionality (dynamic block connections) can somehow be built into the newsched effort (looking @mormj).

matt-g-87, 2022-03-19:
Thanks for the replies.

It makes sense for a multi-input block to consume inputs equally, given that most of them apply some sort of math or comparison across all the inputs. In this case with the Selector block, the intent is to just "get rid of" the unselected input data.

#3762 and #3763 accomodated inputs that had no signals flowing (aka flow rate of 0) whist the selected input does have signals glowing. This is arguably a multi-rate scenario where there are multiple input rates.

More details on my issue:
Right now I'm feeding a Selector block from 4 different Symbol Sync blocks, from 4 different radio channels. The selector is used to output data from the "active" channel, while the other 3 are just tracking noise. When this happens, I believe that the Symbol Sync block is outputting items with very high jitter (as it fails to track a clock from noise), and thus outputs an effective variable rate of data. This can lead to sometimes less/sometimes more samples on the non-selected inputs...and eventual hang if too many samples accumulate on the unselected inputs.

mormj, 2022-03-21:
> I have tested replacing the consume_each(noutput_items) call with the following code with good success:

I could see how this would work in some scenarios, but agree with @willcode that it should be an option.  The default usage of the selector block is to drop inputs on the floor at the same rate as the input being selected.  Certainly wouldn't make the above change the default, but being able to select the input consumption policy makes a lot of sense.  You could have at least 3 options:
1) Consume Inputs Equally
2) Consume Unused Inputs Maximally
3) Don't consume unused inputs

matt-g-87, 2022-04-06:
> I could see how this would work in some scenarios, but agree with @willcode that it should be an option. The default usage of the selector block is to drop inputs on the floor at the same rate as [...]

I like this approach. It satisfies a lot of different use cases.