gnuradio/gnuradio #5654: MacOS M1 chip, flash screen and black screen.

vgegok, 2022-03-20:
### What happened?

I use homebrew to install gnuradio, and source install gr-osmosdr. When I create a project to receive WBFM, My Gnuradio UI will flash or black. At the same time I get some warning in my terminal. But If I delete My grc file, It will work good.

**shutscreen**:
[image]

**grc file**
`options:
  parameters:
    author: vgegok
    catch_exceptions: 'True'
    category: '[GRC Hier Blocks]'
    cmake_opt: ''
    comment: ''
    copyright: ''
    description: ''
    gen_cmake: 'On'
    gen_linking: dynamic
    generate_options: qt_gui
    hier_block_src_path: '.:'
    id: WBFM_Receiver
    max_nouts: '0'
    output_language: python
    placement: (0,0)
    qt_qss_theme: ''
    realtime_scheduling: ''
    run: 'True'
    run_command: '{python} -u {filename}'
    run_options: prompt
    sizing_mode: fixed
    thread_safe_setters: ''
    title: Not titled yet
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [8, 8]
    rotation: 0
    state: enabled

blocks:
- name: freq
  id: variable
  parameters:
    comment: ''
    value: 96.8e6
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [280, 16.0]
    rotation: 0
    state: true
- name: samp_rate
  id: variable
  parameters:
    comment: ''
    value: 2e6
  states:
    bus_sink: false
    bus_source: false
    bus_structure: null
    coordinate: [184, 12]
    rotation: 0
    state: enabled
- name: analog_wfm_rcv_0
  id: analog_wfm_rcv
  parameters:
    affinity: ''
[... 502 lines cut ...]

### Relevant log output

```shell
`% gnuradio-companion 
Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.9.3.0 >>>

Block paths:
	/opt/homebrew/share/gnuradio/grc/blocks

Loading: "/Users/vgegok/VmwareShare/dvbt_tx_hackrf.grc"
>>> Done

Loading: "/Users/vgegok/sdr/gnuradio-project/WBFM_Receiver.grc"
>>> Done

(gnuradio-companion:641): Gtk-WARNING **: 17:02:38.022: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node header owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:641): Gtk-WARNING **: 17:02:38.022: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tabs owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:641): Gtk-WARNING **: 17:02:38.022: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tab owner gnuradio+grc+gui+Notebook+Notebook)

(gnuradio-companion:641): Gtk-WARNING **: 17:02:38.022: Drawing a gadget with negative dimensions. Did you forget to allocate a size? (node tab owner gnuradio+grc+gui+Notebook+Notebook)`
```

marcusmueller, 2022-03-20:
Is this possibly related to #5613 or #5537?

vgegok, 2022-03-20:
Yes, It is same as #5613 !