gnuradio/gnuradio #5664: gr-comedi

kejava, 2022-03-20:
### Feature Description

Hi,
Can gr-comedi be added back into the source?
I looked through the archives and saw that in 2019 it was removed because it's not used.
Hopefully I'm not the only one that has a need for it.
I really like the proposed ability to rapidly throw together a GUI for a bunch of comedi supported hardware.
I just found grc recently and would like to test it out with comedi. I ran through some of the turorials and think it has enormous potential. If something else out there exists, excluding LabVIEW, please let me know. Otherwise grc is the only one that I can find with this kind of flexibilty.
-KevinV

### Feature Urgency

medium (would be nice to have in the near future)

mormj, 2022-03-21:
Hi @kejava - yes it was removed because it had gone unmaintained in the code.  If it is desirable, it would be good to see it ported and maintained as an out of tree module - things don't have to necessarily live in tree to be useful.

kejava, 2022-03-21:
@mormj , ok, i'm not familiar with the OOT modules but i've seen the acronym in the documentation. I'll read up on it and try including it that way. Thanks!

closed 2022-03-21 by kejava

mormj, 2022-03-21:
Sorry for the acronym - basically it just means a module that lives outside of the gnuradio codebase.  https://www.cgran.org/ lists many of the ones that are publicly available