gnuradio/gnuradio #5666: Issues with missing trellis coding python bindings and gui grc blocks for viterbi_combined_xx

bkerler, 2022-03-20:
### What happened?

When trying to port past gr 3.8 oot modules but also when trying to use the trellis viterbi combined module, it seems that the previous trellis viterbi combo modules gr <= 3.8 modules have disappeared (most probably forgotten when porting from swig to pybind11).

When using the viterbi combined module in the gui, I'm getting this error : 

```
Traceback (most recent call last):
  File "/home/hydra/gnuradio/src/broken/gr-dab.trellis_viterbi_broken/examples/top_block.py", line 254, in <module>
    main()
  File "/home/hydra/gnuradio/src/broken/gr-dab.trellis_viterbi_broken/examples/top_block.py", line 232, in main
    tb = top_block_cls()
  File "/home/hydra/gnuradio/src/broken/gr-dab.trellis_viterbi_broken/examples/top_block.py", line 167, in __init__
    self.dab_fic_decode_0 = dab.fic_decode(
  File "/home/hydra/gnuradio/local/lib/python3.10/dist-packages/gnuradio/dab/fic.py", line 92, in __init__
    self.conv_decode = trellis.viterbi_combined_ci(self.fsm, 774, 0, 0,
AttributeError: module 'gnuradio.trellis' has no attribute 'viterbi_combined_ci'
```

The same occurs when trying to port oot modules such as gr-dab or gr-lte, which seem to use : 
```
self.conv_decode = trellis.viterbi_combined_fb(self.fsm, 774, 0, 0, 4, table, trellis.TRELLIS_EUCLIDEAN)
```

Maybe it was transformed into another function ? But then the grc should have been updated as well imho and the oot porting guide doesn't mention it. Other people I've asked seem to have the same issue, that's why they still use gnuradio 3.8. Especially the viterbi_combined_fb module (just just the viterbi_combined_ci) are very important.

Help would be greatly appreciated because then I can port quite some missing (but really important) oot modules to gnuradio 3.10/3.11.

### System Information

OS: Linux Ubuntu 22.04 LTS
GR Installation Method: native distro packages  + Source (3.11 and 3.10 git), gnuradio version 3.10, latest commit

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

v3.10.1.1-54-g594bc948

### Steps to Reproduce the Problem

Just make a new sketch, put a null source to it, add the trellis coding viterbi combo module to it and a null sink. Connect them. Set constellation in the viterbi combo to any value like [1,1,1,1] and dimensionality to any value like 1. Run. Then you will get an error as above.

### Relevant log output

```shell
<<< Welcome to GNU Radio Companion v3.10.1.1-54-g594bc948 >>>

Block paths:
	/home/xxxxxxa/gnuradio/share/gnuradio/grc/blocks

Loading: "/home/xxxxxxa/gnuradio/src/gnuradio/gr-trellis/examples/grc/viterbi_equalization.grc"
>>> Converting from XML
>>> Done

Generating: '/home/xxxxxxa/gnuradio/src/viterbi.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/xxxxxxa/gnuradio/src/viterbi.py

  File "/home/xxxxxxa/gnuradio/src/viterbi.py", line 81
    self.trellis_viterbi_combined_xx_0 = trellis.viterbi_combined_ci(trellis.fsm(), 1, 0, -1, 1, , digital.TRELLIS_EUCLIDEAN)
                                                                                                 ^
SyntaxError: invalid syntax

>>> Done (return code 1)

Generating: '/home/xxxxxxa/gnuradio/src/viterbi.py'
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3 -u /home/xxxxxxa/gnuradio/src/viterbi.py

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Traceback (most recent call last):
  File "/home/xxxxxxa/gnuradio/src/viterbi.py", line 139, in <module>
    main()
  File "/home/xxxxxxa/gnuradio/src/viterbi.py", line 117, in main
    tb = top_block_cls()
  File "/home/xxxxxxa/gnuradio/src/viterbi.py", line 81, in __init__
    self.trellis_viterbi_combined_xx_0 = trellis.viterbi_combined_ci(trellis.fsm(), 1, 0, -1, 1, [1,1,1,1], digital.TRELLIS_EUCLIDEAN)
AttributeError: module 'gnuradio.trellis' has no attribute 'viterbi_combined_ci'

>>> Done (return code 1)
```

closed 2022-03-25 by mormj

n1ai, 2022-07-02:
@bkerler when working on this, did you run into anything that would help explain https://github.com/gnuradio/gnuradio/issues/5985 ?

bkerler, 2022-07-02:
No, I didn't port metrics. Could be that it still needs porting.

PR #5671 (2022-03-25): Add python bindings and qa test for viterbi combined functions
# Pull Request Details
Add missing python bindings for trellis viterbi_combined

## Description
This adds missing python bindings for trellis code, especially viterbi_combinedxx, which seem not to have been introduced after the project conversion from 3.8 (swig) to 3.9 (pybind11)

## Related Issue
Fixes #5666

## Which blocks/areas does this affect?
gr-trellis, in particular all trellis.viterbi_combinedxx python bindings that weren't implemented since porting gnuradio from 3.8 (swig) to 3.9 (pybind11)

## Testing Done
Recompiled gn [... cut]