gnuradio/gnuradio #5695: GNURadio 3.10.1.1 Fails to Build with Custom Source-built Boost 1.78 and UHD 4.1.0.5

icyveins7, 2022-03-27:
### What happened?

I use custom built Boost 1.78.0 in a home folder (not in /usr) instead of the equivalent 'apt install' counterpart. 
Current UHD version is 4.1.0.5, also built from source, using the custom Boost.
The goal is to also build GNURadio into a custom folder.

When building GNURadio 3.10.1.1, I receive an error near the beginning during the make process that it cannot find boost/format.hpp (log is below).

Using GNURadio 3.9.5.0 instead, the build process is successful. All other steps were identical (all dependencies installed, using the same UHD and Boost source builds in their custom folders).

Pointing the CMake to the custom UHD and Boost folders were done successfully (even in the failed 3.10.1.1 version) as both were found and all the functionalities were enabled (e.g. gr-uhd was enabled and all Boost libraries were found).

Hypothesis: This appears to be an error in the CMake generation (Boost libraries/includes in custom folder found but not properly generated into the Makefiles, which GNURadio 3.9.5.0 appears to do correctly).

### System Information

OS: Xubuntu 20.04 LTS
GR Installation Method: Built from source, with Boost 1.78.0 built from source in custom folder, and UHD 4.1.0.5 built from source in custom folder (which also relied on the custom Boost)

### GNU Radio Version

3.10 (maint-3.10)

### Specific Version

3.10.1.1

### Steps to Reproduce the Problem

1) Build Boost 1.78.0 from source to custom folder.
2) Build UHD 4.1.0.5 from source to custom folder, using the custom Boost folder from (1).
3) Build GNURadio from source, using the custom Boost and UHD folder from (1,2).
4) During CMake for GNURadio, specify Boost and UHD custom folders. Configuration and generation is successful.
5) Run `make`, observe error for GNURadio 3.10.1.1 but not 3.9.5.0.

### Relevant log output

```shell
[  0%] Built target doxygen_target

[  0%] Built target gnuradio_docstrings

[  0%] Building CXX object gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch
In file included from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/custom_lock.h:15,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/buffer.h:15,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/buffer_double_mapped.h:15,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/io_signature.h:17,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/basic_block.h:15,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/hier_block2.h:15,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/top_block.h:15,
                 from /home/seoxubuntu/gnuradio-3.10.1.1/build/gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx:5,
                 from <command-line>:
/home/seoxubuntu/gnuradio-3.10.1.1/gnuradio-runtime/include/gnuradio/logger.h:44:10: fatal error: boost/format.hpp: No such file or directory
   44 | #include <boost/format.hpp>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/build.make:64: gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/cmake_pch.hxx.gch] Error 1
make[1]: *** [CMakeFiles/Makefile2:4421: gnuradio-runtime/CMakeFiles/common-precompiled-headers.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

ryanvolz, 2022-03-27:
Looks like the Boost include directory is not added to the `common-precompiled-headers` target, so it would be necessary to add `Boost::format` to https://github.com/gnuradio/gnuradio/blob/maint-3.10/gnuradio-runtime/CMakeLists.txt#L87. And if we've completely removed Boost format on main (not sure if that's done), then this would be a maint-3.10-only change.

mormj, 2022-08-19:
boost::format should be removed completely now on `main` and very minimally included in `logger.h` for `maint-3.10`

b00ga, 2024-01-10:
I'm hitting the same issue on RHEL8/AlmaLinux8 using maint-3.10 branch.  EL8's default boost is 1.66 which is not new enough for the build, and cmake fails appropriately. However, EL8 includes boost 1.69 in parallel installable packages, placing headers under /usr/include/boost169 and libraries under /usr/lib64/boost169.

Setting CMake vars of `-DBoost_INCLUDE_DIR=/usr/include/boost169 -DBoost_LIBRARY_DIR=/usr/lib64/boost169/ ` seems to appropriately detect boost, but I *do not* see `-I/usr/include/boost169` in the failing compile line, so it makes sense that it can't find the offending header.

marcusmueller, 2024-01-31:
Correct, for a self-built Boost, you'll need to `-DBoost_INCLUDE_DIR=…`.

closed 2024-01-31 by marcusmueller