gnuradio/gnuradio #5697: Crashes always in macOS (even when I don't use QT modules)

wlopez-enkoding, 2022-03-28:
### What happened?

Not able to run GNU Radio when using QT modules (I guess)

### System Information

OS: macOS 12.3. 
GR Installation Method: brew

### GNU Radio Version

3.11-git (main)

### Specific Version

0.0.0.0

### Steps to Reproduce the Problem

brew install gnuradio
and then...

installing gr-osmosdr from source code https://github.com/osmocom/gr-osmosdr

marcusmueller, 2022-03-28:
is this x86_64 or an M1 apple?

wlopez-enkoding, 2022-03-28:
> is this x86_64 or an M1 apple?
x86_64 👍

marcusmueller, 2022-03-28:
darn, there goes my theory :smile: 

But that might be a good thing! We've seen similar stuff on M1 and I was assuming it was related to the hardware platform, maybe it isn't.

wlopez-enkoding, 2022-03-30:
> 

Yeah, I looked at all those errors you mentioned, and I have GTK3 versión 3.24.31 installed (port)

wlopez-enkoding, 2022-03-30:
> As I'm not a Mac user myself, all I can do is try and rule out things and narrow down the issue so that when one of our Applexperts come along, they get the best possible start, so: We've seen issues related with display scaling before (#5429), I guess that's not it?
> 
> Since 12.3 is Monterey, maybe it's https://gitlab.gnome.org/GNOME/gtk/-/issues/4342 ; considering [this](https://gitlab.gnome.org/GNOME/gtk/-/issues/4395#note_1309316) comment, is the version of GTK3 high enough (at least 3.24.30_1), as reported by `brew info gtk+3`?

Yeah, I looked at all those errors you mentioned, and I have GTK3 versión 3.24.31 installed (port)

marcusmueller, 2022-03-30:
here comes my OS X ecosystem inexperience: will brew-installed software use ports-installed libraries?

wlopez-enkoding, 2022-04-01:
Guys, I'm closing my use. In my case the solution was the oposite of [this one](https://github.com/gnuradio/gnuradio/issues/5429#issuecomment-1002396426) in #5429 I had to turn scaling off to make it work. Thank you BTW !

closed 2022-04-01 by wlopez-enkoding