gnuradio/gnuradio #5701: single_mapped_buffer with history - incorrect input buffer

AlecVercruysse, 2022-03-29:
### What happened?

Hello, thank you to the devs for your work developing solutions for heterogeneous computing with gnuradio.

Following #5637 to fix the flowgraph freezing issues we were seeing with single mapped buffers, we are able to get a flowgraph running, but we see a discrepancy between the values output by a test copy block with history between a `buffer_double_mapped` and `host_buffer` implementation. Attached are some plots generated by a 
file source -> copy block -> file sink 
flowgraph demonstrating this discrepancy. For the `host_buffer` implementation, the output is nondeterministic.

[image]
[image]

### System Information

OS: Ubuntu 20.04.3 LTS
GR Installation Method: compiled from source

### GNU Radio Version

3.11-git (main)

### Specific Version

v3.11.0.0git-66-g931ae5ff

### Steps to Reproduce the Problem

Attached are the relevant files required to reproduce the problem (provided as .txt files to allow upload to github). The flowgraph is a file source to a copy to a file sink. The block is a simple copy block with `set_history(100)` called, which copies the 100th and on items to the output buffer to try to produce a 1:1 mapping between input and output:

```
  unsigned int offset = HISTORY - 1;
  for (unsigned int n = 0; n < noutput_items; n++) {
      out[n] = in[offset+n];
  }
```

[mwe.py.txt](https://github.com/gnuradio/gnuradio/files/8369173/mwe.py.txt)
[mwe.grc.txt](https://github.com/gnuradio/gnuradio/files/8369174/mwe.grc.txt)
[alecs_block.h.txt](https://github.com/gnuradio/gnuradio/files/8369175/alecs_block.h.txt)
[alecs_block_impl.h.txt](https://github.com/gnuradio/gnuradio/files/8369176/alecs_block_impl.h.txt)
[alecs_block_impl.cc.txt](https://github.com/gnuradio/gnuradio/files/8369177/alecs_block_impl.cc.txt)

marcusmueller, 2022-03-29:
asking @dsorber  to have a look at this, please.

dsorber, 2022-03-30:
Just wanted to mention that I started looking at this but I've run out of time for today.  Unfortunately this is a particularly busy week and I can no longer work on GR stuff while at work.  I'll poke around some more tomorrow evening and in earnest on Sunday, as needed.

AlecVercruysse, 2022-03-31:
Understandable, thank you for your time!

PR #5761 (2022-05-03): runtime: add missing (simulated) data transfer to input/output_blocked_callback functions in host_buffer class
gr-blocks: add `host_buffer_copy` block to be used for testing purposes only

# Pull Request Details

## Description
Data realignment that happens in `single_mapped_buffer`'s input/output blocked callback needs to be propagated to the host or device buffer depending on the buffer's transfer type.  Importantly, this transfer needs to happen atomically inside the input/output blocked callback function.  These cases were accidentally omitted from the `host_buffer` implementation leading  [... cut]
closed 2022-05-03 by mormj