gnuradio/gnuradio #5705: No QT-GUI update from signal source (Brew install, MacOS, M1, Monterey 12.3)

wakass, 2022-03-29:
### What happened?

GUI is not updating after drawing the initial waveform. The GUI itself is perfectly responsive, just the signal doesn't seem to update.

### System Information

OS: Mac,
GR Installation Method: brew

### GNU Radio Version

3.9  (maint-3.9)

### Specific Version

3.9.3.0

### Steps to Reproduce the Problem

Just running a simple signal source, throttle and QT Gui Time sink. I have a GUI Range to change frequency of the signal source, but unfortunately the GUI is not updating after drawing the initial waveform. The GUI itself is perfectly responsive, just the signal doesn't seem to update. I know the "simulation" is running, because removing the throttle block indicates a big cpu-spike.

I did implement the fix for the black screen in the companion. But i'm not sure how that would affect the running gnuradio python instance.

wakass, 2022-04-03:
It appears the update period on the qtgui elements is somehow involved. Setting this to a very small number (e.g. 0.000001) does cause the gui to refresh at a reasonable rate.

wakass, 2022-04-03:
Ok I think i figured it out. The [https://eclecticlight.co/2020/11/27/inside-m1-macs-time-and-logs/](M1's timebase) is different and is off by a factor 41 times from intel. The [high_res_timer ](https://github.com/gnuradio/gnuradio/blob/b554eeab2e70cad520d85dd190eac862cefc9bd0/gnuradio-runtime/include/gnuradio/high_res_timer.h#L93 )for the mach kernel /does/ anticipate this, however i'm afraid the numerator and denominator are inverted in the _ticks per second function. This ofcourse causes a factor of ~40*40 difference from the real timebase. 

```
inline gr::high_res_timer_type gr::high_res_timer_tps(void)
{
    mach_timebase_info_data_t info;
    mach_timebase_info(&info);
    return gr::high_res_timer_type(info.numer * 1000000000UL) / info.denom;
}
```

mormj, 2022-08-19:
Fixed with #5740 (thanks @wakass )

closed 2022-08-19 by mormj